OpenKinect/libfreenect2 #988: Protonect.exe not being created

AAcquier, 2018-04-17:
I managed to build the project using the CMake GUI but Protonect.exe did not appeared in the bin folder, the output I got from CMake is as follow:
C:/Users/Owner/Documents/GitHub/libfreenect2/depends/libusb/MS64/dll/libusb-1.0.dll
 C:/Windows/System32/turbojpeg.dll
 C:/Program Files/NVIDIA GPU Computing Toolkit/CUDA/v9.1/lib/x64/OpenCL.lib
 C:/Program Files/NVIDIA GPU Computing Toolkit/CUDA/v9.1/lib/x64/cudart_static.lib
Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
Configurating examples
Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE) 
Using in-tree freenect2 target
Could NOT find GLFW3 (missing: GLFW3_LIBRARIES) 
Found OpenNI2: C:/Program Files/OpenNI2/Lib  
Feature list:
  CUDA    yes
  CXX11    disabled
  Examples    yes
  OpenCL    yes
  OpenGL    no
  OpenNI2    yes
  TegraJPEG    no
  Threading    tinythread
  TurboJPEG    yes
  VAAPI    no
  VideoToolbox    no (Apple only)
  streamer_recorder    disabled
Configuring done
CMake Warning in CMakeLists.txt:
  **_The build directory is a subdirectory of the source directory.

  This is not supported well by Eclipse.  It is strongly recommended to use a
  build directory which is a sibling of the source directory._**

Generating done

I tried to use the command line but every time I tried it returned:
"'cmake' is not recognized as an internal or external command"
even through I tried to link it to the path of cmake on my machine.

Thanks in advance

SirDifferential, 2018-04-18:
Cmake does not build the project as it's a tool used for creating a Makefile or Visual Studio project specific for your system. After you run cmake you will need to go in the output directory you specified in cmake-gui, then open libfreenect2.sln and use Visual Studio to build the project. Alternatively follow the [readme](https://github.com/OpenKinect/libfreenect2/blob/master/README.md) and do this from the command line.

AAcquier, 2018-04-18:
No  libfreenect2.sln either, I tried the readme also and if you look the end of the first post every time I tired I got "'cmake' is not recognized as an internal or external command" as feedback

closed 2018-04-25 by AAcquier