OpenKinect/libfreenect2 #989: Java binder

AAcquier, 2018-04-25:
I am new to programming but  i know a bit of java an d I would like to know what are the options for the java wrapper, are the wrappers from libfreenect1 will do the trick else can someone point me out in the right direction

floe, 2018-04-26:
See https://github.com/OpenKinect/libfreenect2/issues/417 - we need new Java bindings, SWIG would be one way to generate those. But that's still a TODO.

smokhov, 2018-04-26:
On Thu, 26 Apr 2018, Florian Echtler wrote:

> Date: Thu, 26 Apr 2018 06:07:30 -0700
> 
> See https://github.com/OpenKinect/libfreenect2/issues/417 - we need new 
> Java bindings, SWIG would be one way to generate those. But that's still 
> a TODO.

In the meantime you can use Daniel Shiffman's OpenKinect-for-Processing's 
Java bindings that have been around for a while...

-- 
Serguei Mokhov, PhD, Manager, Networks & Security, AITS | /~\    The ASCII
Affiliate @ Computer Science and Software Engineering & | \ / Ribbon Campaign
Concordia Institute for Information Systems Engineering |  X    Against HTML
Concordia University, Montreal, Quebec, Canada          | / \      Email!
http://www.cs.concordia.ca/~mokhov
---
Co-founder, CCIFF, http://cciff.ca
ET Summit: http://cciff.ca/2017/festival/et/
https://www.youtube.com/channel/UCcJ-jSP6Cya8o-aWoB4l3Pw
---
Technical Lead
http://mdreams-stage.com
http://users.encs.concordia.ca/~gipsy
http://marf.sf.net | http://sf.net/projects/marf

floe, 2018-04-26:
> In the meantime you can use Daniel Shiffman's OpenKinect-for-Processing's Java bindings that have been around for a while...
Excellent, thanks, didn't know these support libfreenect2.

AAcquier, 2018-04-26:
I do not have the skills or the time to generate it myself, I reckon I might be looking into doing in Python that I will have to learn. I am looking at frrenect2, pyfreenect2 and pylibfreenect2  for the wrappers, is there one better or easier to use than the other.

AAcquier, 2018-04-26:
I have been looking into that as well as Java 4 Kinect and they are fairly slow, also I may have to use AI classifier, then maybe  at the end Python might be better

smokhov, 2018-04-26:
On Thu, 26 Apr 2018, AAcquier wrote:

> I have been looking into that as well as Java 4 Kinect and they are 
> fairly slow, also I may have to use AI classifier, then maybe at the end 
> Python might be better

Naive misconception that Python "is faster" than Java run-time 
performance wise.

-- 
Serguei Mokhov, PhD, Manager, Networks & Security, AITS | /~\    The ASCII
Affiliate @ Computer Science and Software Engineering & | \ / Ribbon Campaign
Concordia Institute for Information Systems Engineering |  X    Against HTML
Concordia University, Montreal, Quebec, Canada          | / \      Email!
http://www.cs.concordia.ca/~mokhov
---
Co-founder, CCIFF, http://cciff.ca
ET Summit: http://cciff.ca/2017/festival/et/
https://www.youtube.com/channel/UCcJ-jSP6Cya8o-aWoB4l3Pw
---
Technical Lead
http://mdreams-stage.com
http://users.encs.concordia.ca/~gipsy
http://marf.sf.net | http://sf.net/projects/marf