OpenKinect/libfreenect2 #992: libfreenect2 docker compatibility

itabr, 2018-05-10:
I was wondering if anybody tried using libfreenect2 within docker, I am trying to use libfreenect2 within docker, i am using this component outside the docker and it works correctly, but within docker container it fails so when I tried to test using nucleardreamer/libfreenect2, so i ran 

```
docker run -it \
   -e DISPLAY=:0 \
   -v /tmp/.X11-unix:/tmp/.X11-unix \
   --device /dev/bus/usb:/dev/bus/usb \
   nucleardreamer/libfreenect2 kinect_test 
```
but i got this error : 

```
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: kinect_test [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 17 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:6 with serial 022099999
[Info] [Freenect2Impl] found 1 devices
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
[Error] [OpenGLDepthPacketProcessorImpl] GLFW error 65544 Linux: Failed to watch for joystick connections in /dev/input: No such file or directory
[Error] [OpenGLDepthPacketProcessorImpl] GLFW error 65544 Linux: Failed to open joystick device directory /dev/input: No such file or directory
[Error] [OpenGLDepthPacketProcessorImpl] GLFW error 65543 GLX: Failed to create context: BadMatch (invalid parameter attributes)
[Error] [OpenGLDepthPacketProcessor] Failed to create opengl window.
```
I was wondering if there is any specific configuration that needed to be done to use libfreenect2 within container.

floe, 2018-05-10:
Well, obviously your Docker container doesn't support OpenGL acceleration. You can try if OpenCL works instead (see command line), or find out how to enable OpenGL within Docker.

xlz, 2018-05-13:
Looks like /dev/input is not available in your docker? Try to make it available then.

smokhov, 2018-05-14:
If I were to run libfreenect2 within docker container, I'd run it headless to collect frames and do whatever processing need and ship the frames elsewhere... Why need to run the viewer within the container? `-noviewer` seems like is all what's needed for a successful test.

itabr, 2018-05-16:
thanks for responding my image problem was with OpenCL installation so OpenCL was not working properly within my container, which was solved by using a different base image.

closed 2018-05-16 by itabr