OpenKinect/libfreenect2 #993: LIBUSB_ERROR_NO_DEVICE No such device (it may have been disconnected)

hamsterasu, 2018-05-11:
Hi,

I used to be able to run ```./bin/Protonect``` and it always worked fine. However, while doing a camera calibration using this [ROS Package](https://github.com/code-iai/iai_kinect2/tree/master/kinect2_calibration), the process hanged and after restarting the process, I was never able to get any data from the cameras again. I initially thought it was a problem with that ROS package but when I tried to run ```./bin/Protonect``` , I realised that this driver was the one causing the problem. The viewer would appear but it would be black and after a while it freezes. Below are the messages I get from running the command ```./bin/Protonect```:

```
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_0_39
libva info: va_openDriver() returns 0
[Info] [VaapiRgbPacketProcessorImpl] driver: Intel i965 driver for Intel(R) Skylake - 1.7.0
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 8 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:9 with serial 000932764947
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9729
[Debug] [Freenect2DeviceImpl] status 0x090000: 9731
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 000932764947
device firmware: 4.0.3911.0
[Info] [VaapiRgbPacketProcessor] avg. time: 6.49103ms -> ~154.059Hz
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NO_DEVICE No such device (it may have been disconnected)
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NO_DEVICE No such device (it may have been disconnected)
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NO_DEVICE No such device (it may have been disconnected)
... and more of the same message
```

Below are the relevant logs:

**uname -a**
```
Linux hansel 4.13.0-41-generic #46~16.04.1-Ubuntu SMP Thu May 3 10:06:43 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
```

**lsusb -t**
```
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/6p, 5000M
    |__ Port 1: Dev 5, If 0, Class=Hub, Driver=hub/1p, 5000M
        |__ Port 1: Dev 10, If 0, Class=Vendor Specific Class, Driver=, 5000M
        |__ Port 1: Dev 10, If 1, Class=Vendor Specific Class, Driver=, 5000M
        |__ Port 1: Dev 10, If 2, Class=Audio, Driver=snd-usb-audio, 5000M
        |__ Port 1: Dev 10, If 3, Class=Audio, Driver=snd-usb-audio, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/12p, 480M
    |__ Port 1: Dev 6, If 0, Class=Hub, Driver=hub/1p, 480M
    |__ Port 3: Dev 2, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 3: Dev 2, If 1, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 6: Dev 4, If 0, Class=Video, Driver=uvcvideo, 480M
    |__ Port 6: Dev 4, If 1, Class=Video, Driver=uvcvideo, 480M
    |__ Port 8: Dev 5, If 1, Class=Wireless, Driver=btusb, 12M
    |__ Port 8: Dev 5, If 0, Class=Wireless, Driver=btusb, 12M
```

**lspci -nn**
```
00:00.0 Host bridge [0600]: Intel Corporation Sky Lake Host Bridge/DRAM Registers [8086:1904] (rev 08)
00:02.0 VGA compatible controller [0300]: Intel Corporation Sky Lake Integrated Graphics [8086:1916] (rev 07)
00:04.0 Signal processing controller [1180]: Intel Corporation Skylake Processor Thermal Subsystem [8086:1903] (rev 08)
00:14.0 USB controller [0c03]: Intel Corporation Sunrise Point-LP USB 3.0 xHCI Controller [8086:9d2f] (rev 21)
00:14.2 Signal processing controller [1180]: Intel Corporation Sunrise Point-LP Thermal subsystem [8086:9d31] (rev 21)
00:15.0 Signal processing controller [1180]: Intel Corporation Sunrise Point-LP Serial IO I2C Controller [8086:9d60] (rev 21)
00:15.1 Signal processing controller [1180]: Intel Corporation Sunrise Point-LP Serial IO I2C Controller [8086:9d61] (rev 21)
00:16.0 Communication controller [0780]: Intel Corporation Sunrise Point-LP CSME HECI [8086:9d3a] (rev 21)
00:17.0 SATA controller [0106]: Intel Corporation Sunrise Point-LP SATA Controller [AHCI mode] [8086:9d03] (rev 21)
00:1c.0 PCI bridge [0604]: Intel Corporation Device [8086:9d10] (rev f1)
00:1c.4 PCI bridge [0604]: Intel Corporation Sunrise Point-LP PCI Express Root Port [8086:9d14] (rev f1)
00:1c.5 PCI bridge [0604]: Intel Corporation Sunrise Point-LP PCI Express Root Port [8086:9d15] (rev f1)
00:1f.0 ISA bridge [0601]: Intel Corporation Sunrise Point-LP LPC Controller [8086:9d48] (rev 21)
00:1f.2 Memory controller [0580]: Intel Corporation Sunrise Point-LP PMC [8086:9d21] (rev 21)
00:1f.3 Audio device [0403]: Intel Corporation Sunrise Point-LP HD Audio [8086:9d70] (rev 21)
00:1f.4 SMBus [0c05]: Intel Corporation Sunrise Point-LP SMBus [8086:9d23] (rev 21)
01:00.0 3D controller [0302]: NVIDIA Corporation GM108M [GeForce 940M] [10de:1347] (rev a2)
02:00.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller [10ec:8168] (rev 10)
03:00.0 Network controller [0280]: Intel Corporation Wireless 7265 [8086:095a] (rev 59)
```

**dmesg**
[dmesg.txt](https://github.com/OpenKinect/libfreenect2/files/1996336/dmesg.txt)

When I run ```./bin/Protonect``` with the -nodepth option, I was able to get the video streaming in the viewer for around 10 seconds before it freezes too and the same error messages would appear. 

Also, there was this one time when I tried to run ```./bin/Protonect``` and I was able to get images in the viewer. This was after I stopped trying to run the Kinect for about an hour or so. However, after I stopped the process and tried to run it again, I was back to having the same error messages and I was never able to get it working again. I don't know if it could be a loose connection although I have already tried to replug the Kinect and checked that all the hardware are connected tightly. 

I am running this on **Ubuntu 16.04 LTS with Intel® HD Graphics 520 (Skylake GT2)**

xlz, 2018-05-13:
> I used to be able to

What changed?

I didn't find anything suspicious. I'm seeing some resets in dmesg. Then try replace `true` with `false` in this line https://github.com/OpenKinect/libfreenect2/blob/master/src/libfreenect2.cpp#L1089 and see what happens. (No guarantee though)

hamsterasu, 2018-05-17:
Sorry for the late reply. It turns out that the Kinect was actually faulty and the previous user did not tell me about it. Thank you and so sorry for the trouble!

closed 2018-05-17 by hamsterasu

XunshanMan, 2021-04-29:
Hi I met the same problem and solved it but cost several hours, so I'll leave a message here to help others.

My lsusb shows:

> ➜  build git:(master) lsusb
Bus 002 Device 059: ID 045e:02c4 Microsoft Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 1a81:2038 Holtek Semiconductor, Inc. 
Bus 001 Device 002: ID 046d:c077 Logitech, Inc. M105 Optical Mouse
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

When I use ./bin/Protonect -nodepth, several frames show out and then it freezes.

My dmsg shows:

> [ 4801.174229] xhci_hcd 0000:00:14.0: ERROR Transfer event TRB DMA ptr not part of current TD ep_index 8 comp_code 4
[ 4801.174242] xhci_hcd 0000:00:14.0: Looking for event-dma 000000041546dde0 trb-start 000000041546dc10 trb-end 000000041546dc20 seg-start 000000041546d000 seg-end 000000041546dff0
[ 4801.177948] xhci_hcd 0000:00:14.0: ERROR Transfer event TRB DMA ptr not part of current TD ep_index 8 comp_code 4
[ 4801.177959] xhci_hcd 0000:00:14.0: Looking for event-dma 000000041546ddf0 trb-start 000000041546dc10 trb-end 000000041546dc20 seg-start 000000041546d000 seg-end 000000041546dff0
[ 4801.178193] xhci_hcd 0000:00:14.0: ERROR Transfer event TRB DMA ptr not part of current TD ep_index 8 comp_code 4
[ 4801.178200] xhci_hcd 0000:00:14.0: Looking for event-dma 000000041546de00 trb-start 000000041546dc10 trb-end 000000041546dc20 seg-start 000000041546d000 seg-end 000000041546dff0
[ 4801.181726] xhci_hcd 0000:00:14.0: ERROR Transfer event TRB DMA ptr not part of current TD ep_index 8 comp_code 4
[ 4801.181737] xhci_hcd 0000:00:14.0: Looking for event-dma 000000041546de10 trb-start 000000041546dc10 trb-end 000000041546dc20 seg-start 000000041546d000 seg-end 000000041546dff0

And here is the answer:

The front 3.0 USB port of my laptop is not worked. After I changed to the back 3.0 port everything is well now. 

It seems funny. Hope it may help. Thanks.