OpenKinect/libfreenect2 #1004: kinect run error:Wrong JPEG library version

cyc790, 2018-06-22:
Overview Description:

Ubantu16.04
whem executing ../bin/Protonect, it shows Wrong JPEG library version: library is 90, caller expects 80.
how to solve it?

cyc790, 2018-06-22:
@floe thanks for your answer.I have tried rebuild,but doesn't work

floe, 2018-06-22:
Also when you remove the `build` directory and re-run CMake?

cyc790, 2018-06-22:
@floe yes,I have re-install it.

floe, 2018-06-22:
Hm, can you post the output of `dpkg -l libturbojpeg* libjpeg*`? This is sort of a duplicate of #908, it seems...

cyc790, 2018-06-22:
@floe  OK
 **the output of dpkg -l libturbojpeg* libjpeg*:**
ii  libjpeg-dev:am 8c-2ubuntu8  amd64        Independent JPEG Group's JPEG run
ii  libjpeg-turbo8 1.4.2-0ubunt amd64        IJG JPEG compliant runtime librar
ii  libjpeg-turbo8 1.4.2-0ubunt amd64        Development files for the IJG JPE
un  libjpeg62-dev  <无>         <无>         (无可用描述)
un  libjpeg7-dev   <无>         <无>         (无可用描述)
ii  libjpeg8:amd64 8c-2ubuntu8  amd64        Independent JPEG Group's JPEG run
ii  libjpeg8-dev:a 8c-2ubuntu8  amd64        Independent JPEG Group's JPEG run
ii  libturbojpeg:a 1.4.2-0ubunt amd64        IJG JPEG compliant runtime librar
**and the output of ldd ./bin/Protonect:**
	linux-vdso.so.1 =>  (0x00007ffd795d7000)
	libfreenect2.so.0.2 => /home/cyc/Software/libfreenect2/build/lib/libfreenect2.so.0.2 (0x00007fe983650000)
	libglfw.so.3 => /usr/lib/x86_64-linux-gnu/libglfw.so.3 (0x00007fe98343a000)
	libGL.so.1 => /usr/lib/nvidia-384/libGL.so.1 (0x00007fe9830f8000)
	libstdc++.so.6 => /home/cyc/anaconda3/lib/libstdc++.so.6 (0x00007fe982dbe000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fe9829f4000)
	libusb-1.0.so.0 => /lib/x86_64-linux-gnu/libusb-1.0.so.0 (0x00007fe9827dc000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fe9825bf000)
	libva-drm.so.1 => /usr/lib/x86_64-linux-gnu/libva-drm.so.1 (0x00007fe9823bc000)
	libva.so.1 => /usr/lib/x86_64-linux-gnu/libva.so.1 (0x00007fe9821a0000)
	libjpeg.so.9 => /home/cyc/anaconda3/lib/libjpeg.so.9 (0x00007fe981f64000)
	libturbojpeg.so.0 => /usr/lib/x86_64-linux-gnu/libturbojpeg.so.0 (0x00007fe981d03000)
	libOpenCL.so.1 => /usr/local/cuda-8.0/lib64/libOpenCL.so.1 (0x00007fe981afd000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fe9818f9000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007fe9816f1000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fe9813e8000)
	libX11.so.6 => /usr/lib/x86_64-linux-gnu/libX11.so.6 (0x00007fe9810ae000)
	libXrandr.so.2 => /usr/lib/x86_64-linux-gnu/libXrandr.so.2 (0x00007fe980ea3000)
	libXinerama.so.1 => /usr/lib/x86_64-linux-gnu/libXinerama.so.1 (0x00007fe980ca0000)
	libXi.so.6 => /usr/lib/x86_64-linux-gnu/libXi.so.6 (0x00007fe980a90000)
	libXxf86vm.so.1 => /usr/lib/x86_64-linux-gnu/libXxf86vm.so.1 (0x00007fe98088a000)
	libXcursor.so.1 => /usr/lib/x86_64-linux-gnu/libXcursor.so.1 (0x00007fe980680000)
	libnvidia-tls.so.384.130 => /usr/lib/nvidia-384/tls/libnvidia-tls.so.384.130 (0x00007fe98047c000)
	libnvidia-glcore.so.384.130 => /usr/lib/nvidia-384/libnvidia-glcore.so.384.130 (0x00007fe97e5c0000)
	libXext.so.6 => /usr/lib/x86_64-linux-gnu/libXext.so.6 (0x00007fe97e3ae000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fe983948000)
	libgcc_s.so.1 => /home/cyc/anaconda3/lib/libgcc_s.so.1 (0x00007fe97e19c000)
	libudev.so.1 => /lib/x86_64-linux-gnu/libudev.so.1 (0x00007fe983afb000)
	libdrm.so.2 => /usr/lib/x86_64-linux-gnu/libdrm.so.2 (0x00007fe97df8b000)
	libxcb.so.1 => /home/cyc/anaconda3/lib/libxcb.so.1 (0x00007fe97dd62000)
	libXrender.so.1 => /usr/lib/x86_64-linux-gnu/libXrender.so.1 (0x00007fe97db58000)
	libXfixes.so.3 => /usr/lib/x86_64-linux-gnu/libXfixes.so.3 (0x00007fe97d952000)
	libXau.so.6 => /usr/lib/x86_64-linux-gnu/libXau.so.6 (0x00007fe97d74e000)

thanks for your kindness very much!

floe, 2018-06-22:
I guess the problem is here:
```
 libjpeg.so.9 => /home/cyc/anaconda3/lib/libjpeg.so.9 (0x00007fe981f64000)
```
So anaconda has a newer libjpeg9 installed, which gets picked up at runtime, but at build time, the system library libjpeg8 gets used. You need to remove the anaconda directory from your runtime environment when you want to run Protonect.

floe, 2018-06-22:
(or alternatively downgrade the one installed by anaconda to libjpeg8).

liuhang0727, 2018-08-16:
Hello, i have got the same problem, but i don't know how to "downgrade the one installed by anaconda to libjpeg8", could you give me some advice, thanks.

floe, 2018-08-16:
I don't use anaconda myself, so I don't know. Look in the anaconda docs.

JoeNatan30, 2018-08-18:
Hello, you can downgrade libjpeg in anaconda with this:
``conda install jpeg=8
Remember to rebuild libfreenect2.

Abdul-Mukit, 2018-10-08:
Just wanted to share my experience in case it helps anyone. I am on ubuntu 16 and using Virtualenv with python 2.7. I am not using anaconda3 but it is installed.
What I did was run this command.
`cd
find -name '*libjpeg*'`
and  got:
./anaconda3/envs/ML_Course/lib/libjpeg.so.9.2.0
./anaconda3/envs/ML_Course/lib/libjpeg.so.9
./anaconda3/envs/ML_Course/lib/libjpeg.a
./anaconda3/envs/ML_Course/lib/libjpeg.la
./anaconda3/envs/ML_Course/lib/libjpeg.so
./anaconda3/pkgs/jpeg-9b-h024ee3a_2/lib/libjpeg.so.9.2.0
./anaconda3/pkgs/jpeg-9b-h024ee3a_2/lib/libjpeg.so.9
./anaconda3/pkgs/jpeg-9b-h024ee3a_2/lib/libjpeg.a
./anaconda3/pkgs/jpeg-9b-h024ee3a_2/lib/libjpeg.la
./anaconda3/pkgs/jpeg-9b-h024ee3a_2/lib/libjpeg.so
./anaconda3/lib/libjpeg.so.9.2.0
./anaconda3/lib/libjpeg.so.9
./anaconda3/lib/libjpeg.a
./anaconda3/lib/libjpeg.la
./anaconda3/lib/libjpeg.so

I also noticed this when I ran cmake command:
-- Found JPEG: /home/mukit/anaconda3/lib/libjpeg.so 

I didn't want libjpeg to be taken from anaconda. Also as I find anaconda very confusing I removed anaconda3 completely. Then I went through the installation process again and ran Protonect. This time it worked. I don't know what I did but it worked! :)

ZJU-PLP, 2018-11-15:
@liuhang0727  You can delete all the libjpeg.so 9.2.0  in anaconda3, I solved the problem by this way.And it is essential to show my gratitude to @floe .

closed 2018-11-18 by xlz