OpenKinect/libfreenect2 #1008: Undefined reference errors when running own programs with libfreenect2

smacinnes, 2018-06-29:
I would like to use libfreenect2 to write my own programs with a Kinect v2 sensor. The Protonect example works but that's about it.  I tried to make my own program by following [this](https://openkinect.github.io/libfreenect2/index.html) guide. When I try to compile it I get a bunch of errors (11 to be exact) similar to this:
`kinect1.cpp:(.text+0xa4): undefined reference to libfreenect2::Freenect2::enumerateDevices()`
Before getting this error: 
```
collect2: error: ld returned 1 exit status
<builtin>: recipe for target 'kinect1' failed
make: *** [kinect1] Error 1
```
The enumerateDevices() function and other functions it can't find are stored in [libfreenect2.hpp](https://github.com/OpenKinect/libfreenect2/blob/master/include/libfreenect2/libfreenect2.hpp) and [frame_listener_impl.h](https://github.com/OpenKinect/libfreenect2/blob/master/include/libfreenect2/frame_listener_impl.h), two of the files the [guide](https://openkinect.github.io/libfreenect2/index.html) said to include. Except the functions are only declared; there are no actual function definitions. None of the functions declared in those files have definitions. This would be the reason for the undefined reference errors. 

What am I missing? Where are the definitions for those functions? I know it's probably some stupid little thing I've missed but I'm pulling my hair out over here. I've included the program I was trying to run if you want to see for yourself. There are probably other issues with it that I haven't discovered yet.  Any help on this matter would be greatly appreciated.

[kinect1.txt](https://github.com/OpenKinect/libfreenect2/files/2151152/kinect1.txt)

xlz, 2018-07-02:
https://github.com/OpenKinect/libfreenect2/blob/f46fa65e3a11ee408d43df195c1f837b49358ebb/src/libfreenect2.cpp#L1061

You probably didn't link libfreenect2 to your binary.

AndrewMcClelland, 2019-01-17:
Found a fix using [this answer](https://stackoverflow.com/a/37566893) . Not clear yet to me how that fixed it but I will look into it and see if I can give clearer solution as to why it fixed it for me.

xlz, 2019-01-18:
You have libfreenect2.so in /usr/local/lib, why is that? You copied it yourself?

If you use CMake, like in the answer you linked, with `find_package(freenect2 REQUIRED)`, you'd do exactly as said in README:

> You need to specify cmake -Dfreenect2_DIR=$HOME/freenect2/lib/cmake/freenect2 for CMake based third-party application to find libfreenect2.

If you use pkg-config, `freenect2.pc` is in the installation too, you only need to point `pkg-config` to it.

You can't copy libraries around and expect gcc to find them.