OpenKinect/libfreenect2 #1014: What kernel for Ubuntu 16.04.5 LTS is suitable with USB3 ASmedia controllers?

aaramirezd, 2018-07-31:
Overview Description:

Currently, the ./Protonect only achieve 15-18Hz provably for USB 3.0 driver incompatibility
What is the kernel for Ubuntu 16.04.5 LTS suitable with USB3 ASmedia controllers?
Thanks
Andrés

1. `uname -a`
Linux doctoradob 4.15.0-29-generic #31~16.04.1-Ubuntu SMP Wed Jul 18 08:54:04 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

2. `lsusb -t`
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 10000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/8p, 5000M
    |__ Port 8: Dev 2, If 0, Class=Vendor Specific Class, Driver=, 5000M
    |__ Port 8: Dev 2, If 1, Class=Vendor Specific Class, Driver=, 5000M
    |__ Port 8: Dev 2, If 2, Class=Audio, Driver=snd-usb-audio, 5000M
    |__ Port 8: Dev 2, If 3, Class=Audio, Driver=snd-usb-audio, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/16p, 480M
    |__ Port 7: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 2: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 2: Dev 3, If 1, Class=Human Interface Device, Driver=usbhid, 12M

4. `lspci -nn`
00:00.0 Host bridge [0600]: Intel Corporation Device [8086:591f] (rev 05)
00:01.0 PCI bridge [0604]: Intel Corporation Sky Lake PCIe Controller (x16) [8086:1901] (rev 05)
00:08.0 System peripheral [0880]: Intel Corporation Sky Lake Gaussian Mixture Model [8086:1911]
00:14.0 USB controller [0c03]: Intel Corporation Device [8086:a2af]
00:14.2 Signal processing controller [1180]: Intel Corporation Device [8086:a2b1]
00:16.0 Communication controller [0780]: Intel Corporation Device [8086:a2ba]
00:17.0 SATA controller [0106]: Intel Corporation Device [8086:a282]
00:1b.0 PCI bridge [0604]: Intel Corporation Device [8086:a2e9] (rev f0)
00:1c.0 PCI bridge [0604]: Intel Corporation Device [8086:a297] (rev f0)
00:1f.0 ISA bridge [0601]: Intel Corporation Device [8086:a2c4]
00:1f.2 Memory controller [0580]: Intel Corporation Device [8086:a2a1]
00:1f.3 Audio device [0403]: Intel Corporation Device [8086:a2f0]
00:1f.4 SMBus [0c05]: Intel Corporation Device [8086:a2a3]
01:00.0 VGA compatible controller [0300]: NVIDIA Corporation Device [10de:1b81] (rev a1)
01:00.1 Audio device [0403]: NVIDIA Corporation Device [10de:10f0] (rev a1)
02:00.0 USB controller [0c03]: ASMedia Technology Inc. Device [1b21:2142]
03:00.0 Ethernet controller [0200]: Qualcomm Atheros Device [1969:e0b1] (rev 10)

Steps to Reproduce:

1. doctorado_b@doctoradob:~/usrlib/libfreenect2/build$ ./bin/Protonect cuda

Actual Results:

[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Info] [DepthPacketStreamParser] 30 packets were lost
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Info] [TurboJpegRgbPacketProcessor] avg. time: 56.779ms -> ~17.6121Hz
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Info] [CudaDepthPacketProcessor] avg. time: 1.01754ms -> ~982.76Hz
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
...

xlz, 2018-08-08:
If you have Intel CPU you can use VAAPI.

If USB3 works great it's generally Intel controller. You can use this reference https://unix.stackexchange.com/questions/283296/how-to-find-the-pci-slot-of-an-usb-controller-in-linux to identify working USB controllers.

aaramirezd, 2018-08-09:
Hi @xlz 

In order to use VAAPI, I installed a fresh image of ubuntu 16.04.5 to avoid nvidia-drivers problems.
After, I installed libfreenect2 without cuda support.

However, I have a new problem with VAAPI, see below:

castor@castor-pc:~$ sudo LIBVA_DRIVER_NAME=i965 vainfo
[sudo] password for castor: 
error: XDG_RUNTIME_DIR not set in the environment.
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns 0
libva info: User requested driver 'i965'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_0_39
DRM_IOCTL_I915_GEM_APERTURE failed: Invalid argument
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [22]
param: 4, val: 0
vainfo: intel_memman.c:39: intel_memman_init: Assertion `intel->bufmgr' failed.
Aborted (core dumped)
 
The error with Protonect persists.

Besides, i read the info in this page https://wiki.archlinux.org/index.php/Hardware_video_acceleration#Verification but nothing work.

(Again)

uname -a
Linux castor-pc 4.15.0-30-generic #32~16.04.1-Ubuntu SMP Thu Jul 26 20:25:39 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

lsusb -t
/:  Bus 06.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 5000M
    |__ Port 2: Dev 4, If 0, Class=Vendor Specific Class, Driver=, 5000M
    |__ Port 2: Dev 4, If 1, Class=Vendor Specific Class, Driver=, 5000M
    |__ Port 2: Dev 4, If 2, Class=Audio, Driver=snd-usb-audio, 5000M
    |__ Port 2: Dev 4, If 3, Class=Audio, Driver=snd-usb-audio, 5000M
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 10000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/8p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/16p, 480M
    |__ Port 7: Dev 3, If 0, Class=Hub, Driver=hub/4p, 480M
    |__ Port 9: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 9: Dev 4, If 1, Class=Human Interface Device, Driver=usbhid, 12M

lspci -nn (With the new usb3.0 nec expansion)
00:00.0 Host bridge [0600]: Intel Corporation Device [8086:591f] (rev 05)
00:01.0 PCI bridge [0604]: Intel Corporation Sky Lake PCIe Controller (x16) [8086:1901] (rev 05)
00:08.0 System peripheral [0880]: Intel Corporation Sky Lake Gaussian Mixture Model [8086:1911]
00:14.0 USB controller [0c03]: Intel Corporation Device [8086:a2af]
00:14.2 Signal processing controller [1180]: Intel Corporation Device [8086:a2b1]
00:16.0 Communication controller [0780]: Intel Corporation Device [8086:a2ba]
00:17.0 SATA controller [0106]: Intel Corporation Device [8086:a282]
00:1b.0 PCI bridge [0604]: Intel Corporation Device [8086:a2e9] (rev f0)
00:1c.0 PCI bridge [0604]: Intel Corporation Device [8086:a295] (rev f0)
00:1c.7 PCI bridge [0604]: Intel Corporation Device [8086:a297] (rev f0)
00:1f.0 ISA bridge [0601]: Intel Corporation Device [8086:a2c4]
00:1f.2 Memory controller [0580]: Intel Corporation Device [8086:a2a1]
00:1f.3 Audio device [0403]: Intel Corporation Device [8086:a2f0]
00:1f.4 SMBus [0c05]: Intel Corporation Device [8086:a2a3]
01:00.0 VGA compatible controller [0300]: NVIDIA Corporation Device [10de:1b81] (rev a1)
01:00.1 Audio device [0403]: NVIDIA Corporation Device [10de:10f0] (rev a1)
02:00.0 USB controller [0c03]: ASMedia Technology Inc. Device [1b21:2142]
03:00.0 USB controller [0c03]: NEC Corporation uPD720200 USB 3.0 Host Controller [1033:0194] (rev 03)
04:00.0 Ethernet controller [0200]: Qualcomm Atheros Device [1969:e0b1] (rev 10)

Any suggestion?
Andrés

xlz, 2018-08-10:
You have Sky Lake which is new. Hardware enablement generally is solved by using newer version. You can try pulling libva* from 18.04 https://packages.ubuntu.com/search?keywords=libva-dev. If that doesn't work then you can look at dmesg and see if the kernel reports any errors. Intel/Nvidia dual stack usually works better if "After apt-get install cuda, use sudo prime-select intel to use Intel GPU for desktop."

aaramirezd, 2018-08-10:
Hi @xlz again thanks by the support:

I fonded in dmesg the errors below, I searched in Google but I didn't find anything:
[    3.548103] nouveau 0000:01:00.0: secboot: error during falcon reset: -110
[    3.548106] nouveau 0000:01:00.0: gr: init failed, -110

[    4.930279] nouveau 0000:01:00.0: gr: FECS falcon already acquired by gr!
[    4.930282] nouveau 0000:01:00.0: gr: init failed, -16

I will try with libva from 18.04
Thanks
Andrés

xlz, 2018-08-10:
nouveau is not Nvidia driver. Try to follow the instructions in this README.

closed 2018-11-18 by xlz