OpenKinect/libfreenect2 #1016: Build Error w/ depth_packet_processor

willem0, 2018-08-18:
Steps to Reproduce:

1. Installation from the README.md (usbDK option)
2. cmake .. -G "Visual Studio 15 2017 Win64"
3. cmake --build . --config RelWithDebInfo --target install

I'm getting the following build error for cuda_compile_1_generated_cuda_depth_packet_processor.cu (full log below):

  Building NVCC (Device) object CMakeFiles/cuda_compile_1.dir/src/RelWithDebInfo/cuda_compile_1_generated_cuda_depth_packet_processor.cu.obj

  CMake Error at cuda_compile_1_generated_cuda_depth_packet_processor.cu.obj.RelWithDebInfo.cmake:219 (message):
    Error generating
    C:/Users/will.em/Documents/depth/libfreenect2-master/build/CMakeFiles/cuda_compile_1.dir/src/RelWithDebInfo/cuda_compile_1_generated_cuda_depth_packet_processor.cu.obj

Any suggestions?

---

Full log:

`C:\Users\will.em\Documents\depth2\libfreenect2\build>cmake .. -G "Visual Studio 15 2017 Win64"`
`-- Selecting Windows SDK version 10.0.16299.0 to target Windows 10.0.14393.`
`-- The C compiler identification is MSVC 19.12.25835.0`
`-- The CXX compiler identification is MSVC 19.12.25835.0`
`-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Professional/VC/Tools/MSVC/14.12.25827/bin/Hostx86/x64/cl.exe`
`-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Professional/VC/Tools/MSVC/14.12.25827/bin/Hostx86/x64/cl.exe -- works`
`-- Detecting C compiler ABI info`
`-- Detecting C compiler ABI info - done`
`-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Professional/VC/Tools/MSVC/14.12.25827/bin/Hostx86/x64/cl.exe`
`-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Professional/VC/Tools/MSVC/14.12.25827/bin/Hostx86/x64/cl.exe -- works`
`-- Detecting CXX compiler ABI info`
`-- Detecting CXX compiler ABI info - done`
`-- Detecting CXX compile features`
`-- Detecting CXX compile features - done`
`-- using tinythread as threading library`
`-- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE)`
`-- Found LibUSB: C:/Users/will.em/Documents/depth2/libfreenect2/depends/libusb/MS64/dll/libusb-1.0.lib`
`-- Could NOT find JPEG (missing: JPEG_LIBRARY JPEG_INCLUDE_DIR)`
`-- Could NOT find TegraJPEG (missing: TegraJPEG_LIBRARIES TegraJPEG_INCLUDE_DIRS TegraJPEG_L4T_OK TegraJPEG_DRIVER_OK TegraJPEG_WORKS)`
`-- Performing Test TURBOJPEG_WORKS`
`-- Performing Test TURBOJPEG_WORKS - Success`
`-- Found TurboJPEG: C:/Users/will.em/Documents/depth2/libfreenect2/depends/libjpeg-turbo64/lib/turbojpeg.lib`
`-- Could NOT find GLFW3 (missing: GLFW3_LIBRARIES GLFW3_INCLUDE_DIRS)`
`-- Found OpenGL: opengl32`
`-- Looking for CL_VERSION_2_0`
`-- Looking for CL_VERSION_2_0 - not found`
`-- Looking for CL_VERSION_1_2`
`-- Looking for CL_VERSION_1_2 - found`
`-- Found OpenCL: C:/Program Files/NVIDIA GPU Computing Toolkit/CUDA/v9.2/lib/x64/OpenCL.lib (found version "1.2")`
`-- Found CUDA: C:/Program Files/NVIDIA GPU Computing Toolkit/CUDA/v9.2 (found version "9.2")`
`-- Performing Test COMPILER_HAS_DEPRECATED_ATTR`
`-- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Failed`
`-- Performing Test COMPILER_HAS_DEPRECATED`
`-- Performing Test COMPILER_HAS_DEPRECATED - Success`
`-- Linking with these libraries:`
 `C:/Users/will.em/Documents/depth2/libfreenect2/depends/libusb/MS64/dll/libusb-1.0.lib`
 `C:/Users/will.em/Documents/depth2/libfreenect2/depends/libjpeg-turbo64/lib/turbojpeg.lib`
 `C:/Program Files/NVIDIA GPU Computing Toolkit/CUDA/v9.2/lib/x64/OpenCL.lib`
 `C:/Program Files/NVIDIA GPU Computing Toolkit/CUDA/v9.2/lib/x64/cudart_static.lib`
`-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE)`
`-- Configurating examples`
`-- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE)`
`-- Using in-tree freenect2 target`
`-- Could NOT find GLFW3 (missing: GLFW3_LIBRARIES GLFW3_INCLUDE_DIRS)`
`-- Found OpenNI2: C:/Program Files/OpenNI2/Lib`
`-- Feature list:`
`--   CUDA    yes`
`--   CXX11    disabled`
`--   Examples    yes`
`--   OpenCL    yes`
[... 90 lines cut ...]
    `C:/Users/will.em/Documents/depth2/libfreenect2/build/CMakeFiles/cuda_compile_1.dir/src/RelWithDebInfo/cuda_compile_1_generated_cuda_depth_packet_processor.cu.obj`
``
``
`C:\Program Files (x86)\Microsoft Visual Studio\2017\Professional\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets(171,5): error MSB6006: "cmd.exe" exited with code 1. [C:\Users\will.em\`
`Documents\depth2\libfreenect2\build\freenect2.vcxproj]`
`Done Building Project "C:\Users\will.em\Documents\depth2\libfreenect2\build\freenect2.vcxproj" (default targets) -- FAILED.`
``
`Done Building Project "C:\Users\will.em\Documents\depth2\libfreenect2\build\examples\Protonect.vcxproj" (default targets) -- FAILED.`
``
`Done Building Project "C:\Users\will.em\Documents\depth2\libfreenect2\build\ALL_BUILD.vcxproj" (default targets) -- FAILED.`
``
`Done Building Project "C:\Users\will.em\Documents\depth2\libfreenect2\build\install.vcxproj" (default targets) -- FAILED.`
``
``
`Build FAILED.`
``
`"C:\Users\will.em\Documents\depth2\libfreenect2\build\install.vcxproj" (default target) (1) ->`
`"C:\Users\will.em\Documents\depth2\libfreenect2\build\ALL_BUILD.vcxproj" (default target) (3) ->`
`"C:\Users\will.em\Documents\depth2\libfreenect2\build\examples\Protonect.vcxproj" (default target) (4) ->`
`"C:\Users\will.em\Documents\depth2\libfreenect2\build\freenect2.vcxproj" (default target) (5) ->`
`(CustomBuild target) ->`
  `C:\Program Files (x86)\Microsoft Visual Studio\2017\Professional\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets(171,5): error MSB6006: "cmd.exe" exited with code 1. [C:\Users\will.em\Documents\depth2\libfreenect2\build\freenect2.vcxproj]`
``
    `0 Warning(s)`
    `1 Error(s)`

xlz, 2018-08-19:
There's too little information in the error message unfortunately.

closed 2018-11-18 by xlz

willem0, 2018-11-20:
(For anyone who comes across this thread and is curious, I just switched to Linux, and the install worked fine first try.)