OpenKinect/libfreenect2 #1018: How to detect disconnection of device

jdeeee, 2018-08-28:
Hi!
I am wondering what the best way is to the detect the disconnection of a device at runtime. When I unplug my running device I see an error reading
`[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NO_DEVICE No such device (it may have been disconnected)`
which I think is being output by `libfreenect2::SyncMultiFrameListener::waitForNewFrame`. Unfortunately, I cannot figure out how to detect this error in my code.
Is the best way to detect the disconnection to use `waitForNewFrame` a couple of times and assume the disconnection if no frames are returned (which however would only work as long as there is a timeout)?

Thank you!

PS. I am using https://github.com/code-iai/iai_kinect2/tree/master/kinect2_bridge to run a Kinect v2 in ROS.

SirDifferential, 2018-08-31:
Does libfreenect2 have anything related to device disconnect events?

OpenNI2 does this on Linux by registering for udev events: https://github.com/occipital/OpenNI2/blob/master/ThirdParty/PSCommon/XnLib/Source/Linux/XnLinuxUSB.cpp#L212

On Windows, it uses a callback for [WM_DEVICECHANGE](https://docs.microsoft.com/en-us/windows/desktop/devio/wm-devicechange) at https://github.com/occipital/OpenNI2/blob/master/ThirdParty/PSCommon/XnLib/Source/Win32/XnUSBWin32.cpp#L97

Not sure what they do on Mac though.

xlz, 2018-09-08:
Hotplugging is supported in libusb, which libfreenect2 uses, but it's not supported in libfreenect2.

closed 2018-11-18 by xlz