OpenKinect/libfreenect2 #1024: build make error (Win10 VS2017)

YANGYU106, 2018-10-04:
Hello there,
when I run "cmake --build . --config RelWithDebInfo --target install", there are 2 errors, which are:

c:\users\yangyu_shi\libfreenect2-master\src\libfreenect2.cpp(375): error C2065: 'LIBUSB_OPTION_USE_USBDK': undeclared i
dentifier [C:\Users\Yangyu_Shi\libfreenect2-master\build\freenect2.vcxproj]
c:\users\yangyu_shi\libfreenect2-master\src\libfreenect2.cpp(375): error C3861: 'libusb_set_option': identifier not fou
nd [C:\Users\Yangyu_Shi\libfreenect2-master\build\freenect2.vcxproj]
Done Building Project "C:\Users\Yangyu_Shi\libfreenect2-master\build\freenect2.vcxproj" (default targets) -- FAILED.

Done Building Project "C:\Users\Yangyu_Shi\libfreenect2-master\build\examples\Protonect.vcxproj" (default targets) -- F
AILED.

Done Building Project "C:\Users\Yangyu_Shi\libfreenect2-master\build\ALL_BUILD.vcxproj" (default targets) -- FAILED.

Done Building Project "C:\Users\Yangyu_Shi\libfreenect2-master\build\install.vcxproj" (default targets) -- FAILED.

Build FAILED.

"C:\Users\Yangyu_Shi\libfreenect2-master\build\install.vcxproj" (default target) (1) ->
"C:\Users\Yangyu_Shi\libfreenect2-master\build\ALL_BUILD.vcxproj" (default target) (3) ->
"C:\Users\Yangyu_Shi\libfreenect2-master\build\examples\Protonect.vcxproj" (default target) (4) ->
"C:\Users\Yangyu_Shi\libfreenect2-master\build\freenect2.vcxproj" (default target) (5) ->
(ClCompile target) ->
  c:\users\yangyu_shi\libfreenect2-master\src\libfreenect2.cpp(375): error C2065: 'LIBUSB_OPTION_USE_USBDK': undeclared
 identifier [C:\Users\Yangyu_Shi\libfreenect2-master\build\freenect2.vcxproj]
  c:\users\yangyu_shi\libfreenect2-master\src\libfreenect2.cpp(375): error C3861: 'libusb_set_option': identifier not f
ound [C:\Users\Yangyu_Shi\libfreenect2-master\build\freenect2.vcxproj]

    0 Warning(s)
    2 Error(s)

Could you mind checking this for me? Thanks!

xlz, 2018-10-05:
Your libusb version is too old.

YANGYU106, 2018-10-06:
> Your libusb version is too old.

Thank you so much! I just fixed it! It works now!

closed 2018-10-06 by YANGYU106

mmLukas, 2018-11-14:
Could you please tell me the solution in detail, please? I have wasted a whole day on this issue. Thank you in advance!

xlz, 2018-11-15:
Download the latest libusb.