OpenKinect/libfreenect2 #1027: only get 0 for depth frame data

YANGYU106, 2018-10-09:
Hello there,
I successfully installed libfreenect2 on Win 10 and tested it via Protonect.exe. (OpenGL version 4.5)
However, only "Protonect.exe cpu"can work well. "Protonect.exe gl" can only show rgb image(left down).
Besides, when I run the Python code below: I can only get depth data as 0

```
openni2.initialize()     # can also accept the path of the OpenNI redistribution
dev = openni2.Device.open_any() #register the device
depth_stream = dev.create_depth_stream()    #create the streams
color_stream = dev.create_color_stream()
depth_stream.start()
color_stream.start()

while True:
    dframe = depth_stream.read_frame()
    dframe_data = np.array(dframe.get_buffer_as_triplet()).reshape([480, 640, 2])
    dpt1 = np.asarray(dframe_data[:, :, 0], dtype='float32')
    dpt2 = np.asarray(dframe_data[:, :, 1], dtype='float32')
    dpt2 *= 255
    dpt = dpt1 + dpt2
    cv2.imshow('dpt', dpt)
```

The results of the Protonect.exe test are like below:
```
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: .\install\bin\Protonect.exe [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 11 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:3 with serial 002013164547
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 3*1048576 ir: 8*64*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 002013164547
device firmware: 4.0.3911.0
[Info] [DepthPacketStreamParser] 1 packets were lost
[Info] [OpenGLDepthPacketProcessor] avg. time: 10.3134ms -> ~96.9608Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 15.5862ms -> ~64.1593Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 10.0593ms -> ~99.411Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 15.6363ms -> ~63.9537Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 9.82197ms -> ~101.813Hz
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] already closed, doing nothing

Z:\yshi106\libfreenect2-master\build>.\install\bin\Protonect.exe cpu
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: .\install\bin\Protonect.exe [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 11 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:3 with serial 002013164547
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 3*1048576 ir: 8*64*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 002013164547
device firmware: 4.0.3911.0
[Info] [DepthPacketStreamParser] 30 packets were lost
[Info] [DepthPacketStreamParser] 32 packets were lost
[Info] [DepthPacketStreamParser] 31 packets were lost
[Info] [Freenect2DeviceImpl] stopping...
```
Any idea to help me fix those problems? Thank you!

xlz, 2018-10-09:
Try to build it with CUDA then?

YANGYU106, 2018-10-09:
No depth or IR images either...
Only RGB images(left down) work well. The other three windows are just black.><
```
Z:\yshi106\libfreenect2-master\build>.\install\bin\Protonect.exe cuda
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: .\install\bin\Protonect.exe [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
CUDA pipeline is not supported!
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 11 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:3 with serial 002013164547
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 3*1048576 ir: 8*64*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 002013164547
device firmware: 4.0.3911.0
[Info] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Info] [OpenGLDepthPacketProcessor] avg. time: 9.94782ms -> ~100.525Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 14.9912ms -> ~66.7056Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 9.76118ms -> ~102.447Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 14.6174ms -> ~68.4117Hz
[Info] [Freenect2DeviceImpl] stopping...
```

YANGYU106, 2018-10-10:
I installed libusbk at first, which didn't work.
Then I uninstalled libusbk and installed usbdk to tried it again, which failed to show depth images as well...

YANGYU106, 2018-10-10:
Maybe there is something wrong with the PC itself cause when I changed another pc, all the codes and Protonect.exe worked well... Thank you so much for all the reply!

closed 2018-10-10 by YANGYU106