OpenKinect/libfreenect2 #1034: linked by target "freenect2" in the directory /home/username/libfreenect2

waterbellobin, 2018-10-31:
Overview Description:

Version, Platform, and Hardware Bug Found:

1. `git log -1 --oneline`
2. `uname -a`
3. `lsusb -t`
4. `lspci -nn`

Steps to Reproduce:

1. 
2. 
3. 

Actual Results:

Expected Results:

Reproducibility:

Additional Information:

xlz, 2018-11-03:
What happens if you replace this line

https://github.com/OpenKinect/libfreenect2/blob/master/CMakeLists.txt#L279

with `${OPENGL_LIBRARIES}` and then clean and rebuild?

pfagomez, 2019-08-14:
> What happens if you replace this line
> 
> https://github.com/OpenKinect/libfreenect2/blob/master/CMakeLists.txt#L279
> 
> with `${OPENGL_LIBRARIES}` and then clean and rebuild?

I had the same problem, then I changed the line and I got this problem:

`
cmake ..  -DENABLE_CXX11=ON -DENABLE_OPENCL=ON 
-- The C compiler identification is GNU 7.4.0
-- The CXX compiler identification is GNU 7.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Performing Test LIBFREENECT2_THREADING_STDLIB
-- Performing Test LIBFREENECT2_THREADING_STDLIB - Success
-- using stdlib as threading library
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'libusb-1.0>=1.0.20'
--   Found libusb-1.0, version 1.0.21
-- Checking for modules 'libva;libva-drm'
--   No package 'libva' found
--   No package 'libva-drm' found
-- Found JPEG: /usr/lib/x86_64-linux-gnu/libjpeg.so  
-- Could NOT find TegraJPEG (missing: TegraJPEG_LIBRARIES TegraJPEG_INCLUDE_DIRS TegraJPEG_L4T_OK TegraJPEG_DRIVER_OK TegraJPEG_WORKS) 
-- Performing Test TURBOJPEG_WORKS
-- Performing Test TURBOJPEG_WORKS - Success
-- Found TurboJPEG: /usr/lib/x86_64-linux-gnu/libturbojpeg.so.0  
-- Checking for module 'glfw3'
--   Found glfw3, version 3.2.1
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libOpenGL.so   
-- Looking for CL_VERSION_2_0
-- Looking for CL_VERSION_2_0 - found
-- Found OpenCL: /usr/lib/x86_64-linux-gnu/libOpenCL.so (found version "2.0") 
-- Performing Test OpenCL_C_WORKS
-- Performing Test OpenCL_C_WORKS - Success
-- Performing Test OpenCL_CXX_WORKS
-- Performing Test OpenCL_CXX_WORKS - Success
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found CUDA: /usr/local/cuda (found version "10.1") 
CMake Warning at CMakeLists.txt:391 (MESSAGE):
  Your CUDA installation overrides OpenCL system library path.

-- RPATH set to /usr/lib/x86_64-linux-gnu
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY - Success
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY - Success
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Success
-- Linking with these libraries: 
 /usr/lib/x86_64-linux-gnu/libusb-1.0.so
 /usr/lib/x86_64-linux-gnu/libturbojpeg.so.0
 /usr/lib/x86_64-linux-gnu/libglfw.so
 OPENGL_gl_LIBRARY-NOTFOUND
 /usr/lib/x86_64-linux-gnu/libOpenCL.so
 /usr/local/cuda/lib64/libcudart_static.a
 -lpthread
 dl
 /usr/lib/x86_64-linux-gnu/librt.so
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.13") found components:  doxygen dot 
-- Configurating examples
-- Using in-tree freenect2 target
-- Checking for module 'libopenni2'
--   Found libopenni2, version 2.2.0.3
-- Found OpenNI2: /usr/lib  
-- Feature list:
--   CUDA    yes
--   CXX11    yes
--   Examples    yes
--   OpenCL    yes
--   OpenGL    yes
--   OpenNI2    yes
--   TegraJPEG    no
--   Threading    std::thread
--   TurboJPEG    yes
--   VAAPI    no
--   VideoToolbox    no (Apple only)
--   streamer_recorder    disabled
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
OPENGL_gl_LIBRARY (ADVANCED)
    linked by target "freenect2" in directory /home/user/software/project/libfreenect2
    linked by target "Protonect" in directory /home/user/software/project/libfreenect2/examples

-- Configuring incomplete, errors occurred!
See also "/home/user/software/project/libfreenect2/build/CMakeFiles/CMakeOutput.log".
See also "/home/user/software/project/libfreenect2/build/CMakeFiles/CMakeError.log".

`

KC-FLIR, 2020-04-22:
ok - so fixed this issue on my system, as it was still present 2020/04/22
18.04 Ubuntu, Nvidia 1060 card, nvidia drivers.
Problem was, openGL was just not being found by anything, not just this project.
thanks to [stackoverflow](https://stackoverflow.com/questions/16710047/usr-bin-ld-cannot-find-lnameofthelibrary)  I ran
`ld -lGL --verbose` and of course it couldn't find it, even though glxgears worked just fine and everything else seemed present

Problem? Fix? Super easy.

I had `usr/lib/x86_64-linux-gnu/libGL.so.1` but no `usr/lib/x86_64-linux-gnu/libGL.so` - so softlinked the .so.1 to .so and tada, all fixed and Kinect working.

THANK YOU to all the contributors on this project - and the person who raised this issue. You have my unending gratitude.

M-K21, 2021-05-15:
> ok - so fixed this issue on my system, as it was still present 2020/04/22
> 18.04 Ubuntu, Nvidia 1060 card, nvidia drivers.
> Problem was, openGL was just not being found by anything, not just this [...]

how you have done the softlinked, i'm facing the same problem, but i am using ubuntu 20.04 and kinectv2

pfagomez, 2021-05-17:
> how you have done the softlinked, i'm facing the same problem, but i am using ubuntu 20.04 and kinectv2

ln -s usr/lib/x86_64-linux-gnu/libGL.so.1 usr/lib/x86_64-linux-gnu/libGL.so #(you might need sudo rights)

nworbnhoj, 2022-08-01:
> ok - so fixed this issue on my system, as it was still present 2020/04/22 18.04 Ubuntu, Nvidia 1060 card, nvidia drivers. Problem was, openGL was just not being found by anything, not just this [...]

Thank you - I solved a similar issue with this approach.
Ubuntu 20.04.4 LTS
```
CMake Error at cmake/Dependencies/OpenGL/OpenGL.cmake:15 (message):
  GLU is a dependency of GLEW but was not found.
```
Solved with `sudo ln -s libGLU.so.1 libGLU.so`