OpenKinect/libfreenect2 #1037: Protonect uses wrong OpenGL version (patch included)

cyrozap, 2018-11-03:
Trying to use Protonect with the viewer enabled fails on Mesa 18.1.9 llvmpipe with the following errors:

```
failed to compile vertex shader!
0:1(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.40, 1.00 ES, and 3.00 ES

failed to compile fragment shader!
0:1(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.40, 1.00 ES, and 3.00 ES

failed to link shader program!
error: linking with uncompiled/unspecialized shadererror: linking with uncompiled/unspecialized shader
failed to compile vertex shader!
0:1(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.40, 1.00 ES, and 3.00 ES

failed to compile fragment shader!
0:1(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.40, 1.00 ES, and 3.00 ES

failed to link shader program!
error: linking with uncompiled/unspecialized shadererror: linking with uncompiled/unspecialized shader
```

After discussing this on `#dri-devel`, it seems this is caused by using [`#version 330` shaders](https://github.com/OpenKinect/libfreenect2/blob/a317fc0e1c50f114cd53ec96a137235f9cb5f2af/examples/viewer.cpp#L51) without specifying an OpenGL version that supports them. In this case, [OpenGL 3.1](https://github.com/OpenKinect/libfreenect2/blob/a317fc0e1c50f114cd53ec96a137235f9cb5f2af/examples/viewer.cpp#L33) is specified but [GLSL 3.30 is only supported by OpenGL 3.3 and later](https://en.wikipedia.org/wiki/OpenGL_Shading_Language#Versions).

From what the `#dri-devel` people told me, and from my own experimentation, it seems the proper solutions are to either:
1. Set `GLFW_CONTEXT_VERSION_MINOR` to 3 and `GLFW_OPENGL_PROFILE` to `GLFW_OPENGL_CORE_PROFILE` and keep the shaders as they are.
2. Use whatever OpenGL version we want and rewrite the shaders so they only use features available in the corresponding GLSL version.

Since re-writing shaders and OpenGL code is non-trivial, I've written a simple patch to update the OpenGL version for both Linux and Mac, and have tested it to be working on Linux with the llvmpipe driver:

```diff
diff --git a/examples/viewer.cpp b/examples/viewer.cpp
index c597256..40514be 100644
--- a/examples/viewer.cpp
+++ b/examples/viewer.cpp
@@ -25,14 +25,9 @@ void Viewer::initialize()
     // setup context
     glfwDefaultWindowHints();
     glfwWindowHint(GLFW_CONTEXT_VERSION_MAJOR, 3);
-#ifdef __APPLE__
-    glfwWindowHint(GLFW_CONTEXT_VERSION_MINOR, 2);
+    glfwWindowHint(GLFW_CONTEXT_VERSION_MINOR, 3);
     glfwWindowHint(GLFW_OPENGL_FORWARD_COMPAT, GL_TRUE);
     glfwWindowHint(GLFW_OPENGL_PROFILE, GLFW_OPENGL_CORE_PROFILE);
-#else
-    glfwWindowHint(GLFW_CONTEXT_VERSION_MINOR, 1);
-    glfwWindowHint(GLFW_OPENGL_PROFILE, GLFW_OPENGL_ANY_PROFILE);
-#endif
     //glfwWindowHint(GLFW_VISIBLE, debug ? GL_TRUE : GL_FALSE);
 
     window = glfwCreateWindow(win_width*2, win_height*2, "Viewer (press ESC to exit)", 0, NULL);
```

I'll test the patch later on the i915 driver on Linux and on a Mac and update this issue once I've done so.

xlz, 2018-11-04:
You can't simply change the version numbers, see https://github.com/OpenKinect/libfreenect2/blame/master/examples/viewer.cpp#L29-L33. The version numbers are set the the values for good reasons.

https://github.com/OpenKinect/libfreenect2/commit/2cee7df712be404d940ee9b8f39ce3461000baf5 "It's possible viewer's shader version 330 needs to be ported to version 140, but no bugs were encountered at the moment."

MengHao666, 2020-06-17:
I meet the same problem ,what should I do?