OpenKinect/libfreenect2 #1038: Cmake erros when building

hibetterheyj, 2018-11-11:
Version, Platform, and Hardware Bug Found:

-  `git log -1 --oneline`

`a317fc0 Fix contructor usage in C++98`

-  `uname -a`

`Linux philhe-Ubuntu 4.13.0-45-generic #50~16.04.1-Ubuntu SMP Wed May 30 11:18:27 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux`

-  `lsusb -t`

```
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/8p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/16p, 480M
    |__ Port 1: Dev 8, If 0, Class=Hub, Driver=hub/2p, 480M
        |__ Port 1: Dev 10, If 0, Class=Vendor Specific Class, Driver=, 480M
        |__ Port 2: Dev 11, If 0, Class=Vendor Specific Class, Driver=, 480M
    |__ Port 4: Dev 2, If 1, Class=Video, Driver=uvcvideo, 480M
    |__ Port 4: Dev 2, If 0, Class=Video, Driver=uvcvideo, 480M
    |__ Port 6: Dev 5, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
    |__ Port 9: Dev 4, If 0, Class=Wireless, Driver=btusb, 12M
    |__ Port 9: Dev 4, If 1, Class=Wireless, Driver=btusb, 12M
```

Actual Results:

Dear Developers,

I just follow the installation steps here

my folder is in /home/philhe/Desktop/machine_vision/Hardware/kinect/other_code/libfreenect2 but not home/...

os: ubuntu 16.04

    git clone https://github.com/OpenKinect/libfreenect2.git
    cd libfreenect2
    sudo apt-get install build-essential cmake pkg-config
    sudo apt-get install libusb-1.0-0-dev
    
    sudo apt-get install libturbojpeg libjpeg-turbo8-dev
    sudo apt-get install libglfw3-dev

Here encounters 2 ERRORs

    E: Package 'libturbojpeg' has no installation candidate
    E: Unable to locate package beignet-dev

then I try to build & make

    mkdir build && cd build
    cmake .. -DCMAKE_INSTALL_PREFIX=$HOME/freenect2

when cmake, I encounter many ERRORs

    cmake .. -DCMAKE_INSTALL_PREFIX=$HOME/freenect2
    -- The C compiler identification is GNU 5.4.0
    -- The CXX compiler identification is GNU 5.4.0
    -- Check for working C compiler: /usr/bin/cc
    -- Check for working C compiler: /usr/bin/cc -- works
    -- Detecting C compiler ABI info
    -- Detecting C compiler ABI info - done
    -- Detecting C compile features
    -- Detecting C compile features - done
    -- Check for working CXX compiler: /usr/bin/c++
    -- Check for working CXX compiler: /usr/bin/c++ -- works
    -- Detecting CXX compiler ABI info
    -- Detecting CXX compiler ABI info - done
    -- Detecting CXX compile features
    -- Detecting CXX compile features - done
    -- using tinythread as threading library
    -- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
    -- Checking for module 'libusb-1.0>=1.0.20'
    --   Found libusb-1.0, version 1.0.20
    -- Checking for modules 'libva;libva-drm'
    --   No package 'libva' found
    --   No package 'libva-drm' found
    -- Found JPEG: /usr/lib/x86_64-linux-gnu/libjpeg.so  
    -- Could NOT find TegraJPEG (missing:  TegraJPEG_LIBRARIES TegraJPEG_INCLUDE_DIRS TegraJPEG_L4T_OK TegraJPEG_DRIVER_OK TegraJPEG_WORKS) 
    -- Performing Test TURBOJPEG_WORKS
    -- Performing Test TURBOJPEG_WORKS - Failed
    CMake Error at /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
      Could NOT find TurboJPEG (missing: TURBOJPEG_WORKS)
    Call Stack (most recent call first):
      /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:388 (_FPHSA_FAILURE_MESSAGE)
      cmake_modules/FindTurboJPEG.cmake:66 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
      CMakeLists.txt:242 (FIND_PACKAGE)
    
    
    -- Configuring incomplete, errors occurred!
    See also "/home/philhe/Desktop/machine_vision/Hardware/kinect/other_code/libfreenect2/build/CMakeFiles/CMakeOutput.log".
    See also "/home/philhe/Desktop/machine_vision/Hardware/kinect/other_code/libfreenect2/build/CMakeFiles/CMakeError.log".

I try many times to delete buildfolder but it still cannot work. I think it may be a duplicate of #743

so I just try find /usr/local -name '*libusb*' -delete and reinstall libusb, it cannot work either!

can you give me some advice on it?

Best wishes!

xlz, 2018-11-17:
> Package 'libturbojpeg' has no installation candidate

Add universe to sources.list.

closed 2018-11-18 by xlz

hclin77923, 2019-02-19:
Hello,

I have the similar issue as this post; however, I successfully install 'libturbojpeg' and 'libjpeg-turbo8-dev' by
`sudo apt-get install libturbojpeg libjpeg-turbo8-dev`

However, the error still comes up when I configure cmake.

```
~/libfreenect2/build$ cmake .. -DCMAKE_INSTALL_PREFIX=$HOME/freenect2
-- using stdlib as threading library
-- Could NOT find TegraJPEG (missing:  TegraJPEG_LIBRARIES TegraJPEG_INCLUDE_DIRS TegraJPEG_L4T_OK TegraJPEG_DRIVER_OK TegraJPEG_WORKS) 
CMake Error at /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
  Could NOT find TurboJPEG (missing: TURBOJPEG_WORKS)
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:388 (_FPHSA_FAILURE_MESSAGE)
  cmake_modules/FindTurboJPEG.cmake:66 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:243 (FIND_PACKAGE)

-- Configuring incomplete, errors occurred!
See also "/home/hclin/libfreenect2/build/CMakeFiles/CMakeOutput.log".
See also "/home/hclin/libfreenect2/build/CMakeFiles/CMakeError.log". 
```
I also tried someone use sym-link in the post #36 
`sudo ln -s /usr/lib/x86_64-linux-gnu/libturbojpeg.so.0 /usr/lib/x86_64-linux-gnu/libturbojpeg.so`

It turned out not work, so I removed that sym-link and reinstall libturbojpg back. I don't know how to resolve my current issue. 
Can you please provide some suggestion to resolve it? Thank you!