OpenKinect/libfreenect2 #1040: 'examples/CMakeFiles/Protonect.dir/all' failed

jsdd25, 2018-11-15:
TX2 Ubuntu:16.04 
When I run make command,I encountered the following errors:
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libva-drm.so: undefined reference to `drmGetClient'
collect2: error: ld returned 1 exit status
examples/CMakeFiles/Protonect.dir/build.make:160: recipe for target 'bin/Protonect' failed
make[2]: *** [bin/Protonect] Error 1
CMakeFiles/Makefile2:278: recipe for target 'examples/CMakeFiles/Protonect.dir/all' failed
make[1]: *** [examples/CMakeFiles/Protonect.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

Can anyone help me? Thank you very much!

xlz, 2018-11-17:
Use `cmake .. -DENABLE_VAAPI=OFF`.

closed 2018-11-18 by xlz

jsdd25, 2018-11-19:
Thank you very much! I have fixed this problem.