OpenKinect/libfreenect2 #1043: [Linux 16.04 Jetson TX2] Color Camera Tint?

Calvin315, 2018-11-21:
Overview Description:
Hello,
When I run Protonect, I get a tinted video stream? I am not too sure what is going on with the system. I had some issues with installation of freenect2 with the JetsonHacks guide and am not sure if something went wrong in there. that might have caused this issue. When I use the Kinect with Windows, it properly displays the colored images and a depth stream.

This is the video stream I get from the camera.
[image]

This is my console stream.
    ubuntu@tegra-ubuntu:~/libfreenect2/build/bin$ ./Protonect gl
    Version: 0.2.0
    Environment variables: LOGFILE=<protonect.log>
    Usage: ./Protonect [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
            [-noviewer] [-norgb | -nodepth] [-help] [-version]
            [-frames <number of frames to process>]
    To pause and unpause: pkill -USR1 Protonect
    [Info] [Freenect2Impl] enumerating devices...
    [Info] [Freenect2Impl] 7 usb devices connected
    [Info] [Freenect2Impl] found valid Kinect v2 @2:4 with serial 150274234547
    [Info] [Freenect2Impl] found 1 devices
    [Info] [Freenect2DeviceImpl] opening...
    [Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
    [Info] [Freenect2DeviceImpl] opened
    [Info] [Freenect2DeviceImpl] starting...
    [Debug] [Freenect2DeviceImpl] status 0x090000: 9763
    [Debug] [Freenect2DeviceImpl] status 0x090000: 9763
    [Info] [Freenect2DeviceImpl] submitting rgb transfers...
    [Info] [Freenect2DeviceImpl] submitting depth transfers...
    [Info] [Freenect2DeviceImpl] started
    device serial: 150274234547
    device firmware: 4.0.3917.0
    [Debug] [DepthPacketStreamParser] not all subsequences received 0
    [Debug] [DepthPacketStreamParser] skipping depth packet
    [Debug] [DepthPacketStreamParser] skipping depth packet
    [Debug] [DepthPacketStreamParser] skipping depth packet
    [Info] [DepthPacketStreamParser] 10 packets were lost
    [Debug] [DepthPacketStreamParser] skipping depth packet
    [Debug] [DepthPacketStreamParser] skipping depth packet
    [Debug] [DepthPacketStreamParser] skipping depth packet
    [Info] [DepthPacketStreamParser] 3 packets were lost
    [Info] [OpenGLDepthPacketProcessor] avg. time: 23.4569ms -> ~42.6314Hz
    [Info] [TegraJpegRgbPacketProcessor] avg. time: 3.58839ms -> ~278.676Hz
    [Info] [OpenGLDepthPacketProcessor] avg. time: 22.076ms -> ~45.298Hz
    [Info] [TegraJpegRgbPacketProcessor] avg. time: 3.33384ms -> ~299.955Hz
    [Info] [OpenGLDepthPacketProcessor] avg. time: 23.315ms -> ~42.8909Hz
    [Info] [TegraJpegRgbPacketProcessor] avg. time: 3.36017ms -> ~297.604Hz
    [Info] [OpenGLDepthPacketProcessor] avg. time: 22.0065ms -> ~45.441Hz
    [Info] [TegraJpegRgbPacketProcessor] avg. time: 3.36711ms -> ~296.991Hz
    [Info] [OpenGLDepthPacketProcessor] avg. time: 22.2562ms -> ~44.9314Hz
    [Info] [TegraJpegRgbPacketProcessor] avg. time: 3.38398ms -> ~295.51Hz
    [Debug] [DepthPacketStreamParser] skipping depth packet
    [Info] [DepthPacketStreamParser] 1 packets were lost

Calvin315, 2018-11-23:
Sorry to reopen this but do you also know why it is skipping the depth packet or where the issue for that might be?

leviresende, 2019-05-13:
Hey guys!
I'm using the JETSON TX 2 with the following configuration:
 L4T 28.2.1 [ JetPack 3.3 or 3.2.1 ]
 Board: t186ref
 Ubuntu 16.04.6 LTS
 Kernel Version: 4.4.38-tegra
 CUDA 9.0.252

And I'd like to connect it with a Kinect v2, however, I didn't have success. I've done everything that I read on the forums here on Github and on JetsonHacks, even tried to install as Kinect v1, but nothing. After all, I saw that it's not possible to connect a default board (TX2) with this device. Is it true?

By the way, which camera is it better to develop SLAM applications? ZED?

Thanks in advance.