OpenKinect/libfreenect2 #1045: Representation of DepthPacketStreamParser [error] & [warning]

seanren96, 2018-12-05:
Overview Description:

Hi, I have encountered some error and warning while running /libfreenect2/examples/protonect after I build it today. 
Once the operation starts, there are [error] shows up as : 
    1. [Error] [DepthPacketStreamParser] not all subsequences received 0. 
    2.[Error] [DepthPacketStreamParser] not all subsequences received 896
    3.[Error] [DepthPacketStreamParser] not all subsequences received 895
After these error, the program comes with tons of [warning] as following:
    1.[Warning] [DepthPacketStreamParser] skipping depth packet
    2.[Warning] [RgbPacketStreamParser] skipping rgb packet!
**The goal I want to achieve is getting depth images from kinect v2 -> save the images as 16bits bmp.**
**So far, I want to know what do those error represent also, how to display and save the depth images.** 

Version, Platform, and Hardware Bug **Found:**
1. `git log -1 --oneline`
2. `uname -a`
Linux odroid 3.10.82-56 #1 SMP PREEMPT Wed Sep 2 19:43:47 BRT 2015 armv7l armv7l armv7l GNU/Linux
3. `lsusb -t`
odroid@odroid:~/libfreenect2/examples/protonect/bin$ lsusb -t
/:  Bus 06.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Vendor Specific Class, Driver=r8152, 5000M
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 480M
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/2p, 5000M
        |__ Port 2: Dev 3, If 0, Class=Hub, Driver=hub/1p, 5000M
            |__ Port 1: Dev 4, If 0, Class=Vendor Specific Class, Driver=, 5000M
            |__ Port 1: Dev 4, If 1, Class=Vendor Specific Class, Driver=, 5000M
            |__ Port 1: Dev 4, If 2, Class=Audio, Driver=snd-usb-audio, 5000M
            |__ Port 1: Dev 4, If 3, Class=Audio, Driver=snd-usb-audio, 5000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/2p, 480M
        |__ Port 2: Dev 3, If 0, Class=Hub, Driver=hub/1p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=exynos-ohci/3p, 12M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=s5p-ehci/3p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 2: Dev 6, If 0, Class=Video, Driver=uvcvideo, 480M
        |__ Port 2: Dev 6, If 1, Class=Video, Driver=uvcvideo, 480M
        |__ Port 2: Dev 6, If 2, Class=Audio, Driver=snd-usb-audio, 480M
        |__ Port 2: Dev 6, If 3, Class=Audio, Driver=snd-usb-audio, 480M
        |__ Port 3: Dev 7, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 4: Dev 8, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
4. `lspci -nn`
pcilib: Cannot open /proc/bus/pci
lspci: Cannot find any working access method.
Steps to Reproduce:
1. ./Protonect

Actual Results:
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 16 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:4 with serial 501326440242
[Info] [Freenect2Impl] found 1 devices
OpenGL pipeline is not supported!
[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Info] [OpenCLDepthPacketProcessorImpl]   0: Mali-T628 (GPU)[ARM]
[Info] [OpenCLDepthPacketProcessorImpl]   1: Mali-T628 (GPU)[ARM]
[Info] [OpenCLDepthPacketProcessorImpl] selected device: Mali-T628 (GPU)[ARM]
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] ReadData0x14 response
[Debug] [Freenect2DeviceImpl] 92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00    2e2133552e2e2e202e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 08 12 05 47 4b 52 35 32 31 2e 31    31332e2e2e2e2e2e474b523532312e31
0x0050:  58 00 00 00 00 00 00 00 07 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

[Debug] [Freenect2DeviceImpl] ReadStatus0x090000 response
[Debug] [Freenect2DeviceImpl] 4 bytes of raw data
0x0000:  01 2e 00 00                                        2e2e2e2e

[Debug] [Freenect2DeviceImpl] ReadStatus0x090000 response
[Debug] [Freenect2DeviceImpl] 4 bytes of raw data
0x0000:  03 2e 00 00                                        2e2e2e2e

[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Error] [DepthPacketStreamParser] not all subsequences received 0
[Info] [Freenect2DeviceImpl] started
device serial: 501326440242
device firmware: 4.1.3911.0.7
[Error] [DepthPacketStreamParser] not all subsequences received 896
[Error] [DepthPacketStreamParser] not all subsequences received 895
[Warning] [DepthPacketStreamParser] skipping depth packet
[Warning] [DepthPacketStreamParser] skipping depth packet
[Warning] [DepthPacketStreamParser] skipping depth packet
[Warning] [DepthPacketStreamParser] skipping depth packet
[Warning] [DepthPacketStreamParser] skipping depth packet
[Warning] [DepthPacketStreamParser] skipping depth packet
[Warning] [DepthPacketStreamParser] skipping depth packet
[Warning] [RgbPacketStreamParser] skipping rgb packet!
[Warning] [DepthPacketStreamParser] skipping depth packet
[Warning] [RgbPacketStreamParser] skipping rgb packet!
[Warning] [DepthPacketStreamParser] skipping depth packet
[Warning] [DepthPacketStreamParser] skipping depth packet
[Warning] [RgbPacketStreamParser] skipping rgb packet!
[Warning] [DepthPacketStreamParser] skipping depth packet

Expected Results:

Reproducibility:

Additional Information:

xlz, 2018-12-05:
The errors and warnings are nothing major. They mean some images are lost due to slow USB transfer.

> how to display

Protonect already does.

> and save the depth images

Check https://github.com/OpenKinect/libfreenect2/tree/master/tools/streamer_recorder

seanren96, 2018-12-05:
Thank you for the kind help. I think I might close this thread for now.

closed 2018-12-05 by seanren96