OpenKinect/libfreenect2 #1047: Problem with ./Protonect cpu

GutlapalliNikhil, 2018-12-06:
Hello guys,
After installing libfreenect2, i wanted to test with ./Protonect. So i tried with ./Protonect cl and ./Protonect gl and ./Protonect cpu. When i tested with ./Protonect cl and ./Protonect gl, the video flow is good and fine. And when i tested with ./Protonect cpu, the video flow is pausing and getting warnings. Help me please.

nikhil@nikhil-HP-Notebook-15:~/libfreenect2/build/bin$ ./Protonect cpu
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./Protonect [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 6 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:6 with serial 001409763847
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9729
[Debug] [Freenect2DeviceImpl] status 0x090000: 9731
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 001409763847
device firmware: 4.0.3911.0
[Debug] [DepthPacketStreamParser] not all subsequences received 0
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] not all subsequences received 383
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] not all subsequences received 887
[Debug] [DepthPacketStreamParser] not all subsequences received 991
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] not all subsequences received 1015
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] not all subsequences received 767
[Debug] [DepthPacketStreamParser] not all subsequences received 967
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] not all subsequences received 767
[Info] [DepthPacketStreamParser] 35 packets were lost
[Debug] [DepthPacketStreamParser] not all subsequences received 503
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] not all subsequences received 1015
[Debug] [DepthPacketStreamParser] not all subsequences received 945
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] not all subsequences received 959
[Debug] [DepthPacketStreamParser] not all subsequences received 959
[Debug] [DepthPacketStreamParser] not all subsequences received 767
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] not all subsequences received 959
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] not all subsequences received 959
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] not all subsequences received 991
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] not all subsequences received 511
[Debug] [DepthPacketStreamParser] not all subsequences received 1007
[Debug] [DepthPacketStreamParser] not all subsequences received 511
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] not all subsequences received 1015
[Debug] [DepthPacketStreamParser] not all subsequences received 991
[Debug] [DepthPacketStreamParser] not all subsequences received 1015
[Info] [DepthPacketStreamParser] 35 packets were lost
[Debug] [DepthPacketStreamParser] not all subsequences received 1022
[Debug] [DepthPacketStreamParser] not all subsequences received 1019
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] not all subsequences received 951
[Debug] [DepthPacketStreamParser] not all subsequences received 767
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] not all subsequences received 879
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] not all subsequences received 1022
[Debug] [DepthPacketStreamParser] not all subsequences received 999
[Debug] [DepthPacketStreamParser] not all subsequences received 1021

floe, 2018-12-06:
If the CL and GL parsers work for you, then there is no reason to use the CPU parser, because it's very slow and just a fallback if the other parsers don't work. That is the reason for the warnings.

closed 2018-12-06 by floe

GutlapalliNikhil, 2018-12-06:
Thank you for sharing information. Can you please help me in this...
https://github.com/code-iai/iai_kinect2/issues/518