OpenKinect/libfreenect2 #1048: Release lib for windows not compiled with CUDA or openCL

fred-dev, 2018-12-18:
Overview Description:

Version, Platform, and Hardware Bug Found:
Release 0.2.0 Win10 VS2015/VS2017 Release

1. `git log -1 --oneline`
2. `uname -a`
3. `lsusb -t`
4. `lspci -nn`

Steps to Reproduce:

1. Make a VS project include the headers and lib from the release v0.2.0
add the freenect.lib to the project
properties->linker->input->additional dependancies
and added the path to the lib in
properties->linker->general->additional library directories

Install the nvidia gpu computing toolkit (with examples) and add OpenCL.lib to
properties->linker->input->additional dependancies
and added the path to the lib in
properties->linker->general->additional library directories
and added the opencl include folder to
properties->C/C++->General->additional include directories
2. Complie

Actual Results:
Errors on compilation:

when using LIBFREENECT2_WITH_OPENCL_SUPPORT I get 2 errors:

`Error	unresolved external symbol "public: __cdecl libfreenect2::OpenCLKdePacketPipeline::OpenCLKdePacketPipeline(int)" (??0OpenCLKdePacketPipeline@libfreenect2@@QEAA@H@Z)	example C:\Users\Fred\Documents\openFrameworks\addons\ofxKinectV2\example C:\Users\Fred\Documents\openFrameworks\addons\ofxKinectV2\example\ofProtonect.obj 1`

and

`Error	unresolved external symbol "public: __cdecl libfreenect2::OpenCLPacketPipeline::OpenCLPacketPipeline(int)" (??0OpenCLPacketPipeline@libfreenect2@@QEAA@H@Z)	example C:\Users\Fred\Documents\openFrameworks\addons\ofxKinectV2\example C:\Users\Fred\Documents\openFrameworks\addons\ofxKinectV2\example\ofProtonect.obj 1`

When attempting to switch to LIBFREENECT2_WITH_CUDA_SUPPORT

I get 4 errors:

```
CUDAKDEPacketPipeline': is not a member of 'libfreenect2'	
CUDAPacketPipeline': is not a member of 'libfreenect2'

syntax error: identifier 'CUDAKDEPacketPipeline'	
syntax error: identifier 'CUDAPacketPipeline'
```

Expected Results:
Compilation with the ability to use openCL

Reproducibility:
100%
Additional Information:
It seems like the libs in the releases section are not compiled with openCL or CUDA. I can add them to a project and use opengl or CPU as a packet processor but not any other accelerated libraries like cuda, openCL and turbojpeg. Is this correct? Is it deliberate? Are compiled libs available with CUDA and OpenCL available?

xlz, 2018-12-18:
It's deliberate. CUDA/OpenCL's versions vary too much. When I built it I couldn't guarantee binary compatibility so only OpenGL was built in the binary. You need to build your own libfreenect2 with CUDA turned on etc.

fred-dev, 2018-12-18:
Ok, I managed to do that, I now have access to openCL, which is great, but CUDA still gives errors here are my cmake settings:

```
using stdlib as threading library
Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE) 
Linking with these libraries: 
 C:/Users/Fred/Documents/libfreenect2/depends/libusb_src/x64/Release/lib/libusb-1.0.lib
 C:/Users/Fred/Documents/libfreenect2/depends/libjpeg-turbo64/bin/libturbojpeg.lib
 C:/Users/Fred/Documents/libfreenect2/depends/glfw/lib-vc2015/glfw3dll.lib
 opengl32
 C:/Program Files/NVIDIA GPU Computing Toolkit/CUDA/v10.0/lib/x64/OpenCL.lib
 C:/Program Files/NVIDIA GPU Computing Toolkit/CUDA/v10.0/lib/x64/cudart.lib
Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
Configurating examples
Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE) 
Using in-tree freenect2 target
Feature list:
  CUDA    yes
  CXX11    disabled
  Examples    yes
  OpenCL    yes
  OpenGL    yes
  OpenNI2    disabled
  TegraJPEG    disabled
  Threading    std::thread
  TurboJPEG    yes
  VAAPI    disabled
  VideoToolbox    no (Apple only)
  streamer_recorder    disabled
```

And here are the errors:

```
Error	'CUDAKDEPacketPipeline': is not a member of 'libfreenect2'	
Error	'CUDAPacketPipeline': is not a member of 'libfreenect2'	
Error	syntax error: identifier 'CUDAKDEPacketPipeline'	
Error	syntax error: identifier 'CUDAPacketPipeline'	
```
Am I doing something wrong in the configuration for the build? I don't get errors building the lib. I do notice that if I clean the cmake build folder, build only with opencl enabled and check the file modification date of freenect.lib and then rebuild with cuda enabled freenect.lib is not modified and I would have expected it to be changed. Is this normal?

xlz, 2018-12-19:
Recommend to clean up the build directory and generated files before retrying if cmake config doesn't produce good results.

xlz, 2018-12-19:
Maybe look at the error in more detail instead. Does your https://github.com/OpenKinect/libfreenect2/blob/ae39a6d5dc024a49702ef7cd93b72d6045773fc7/include/libfreenect2/packet_pipeline.h have CUDAPacketPipeline etc.

fred-dev, 2018-12-22:
Rebuilt from scratch and I now have access to the CUDA pipeline.

closed 2018-12-22 by fred-dev