OpenKinect/libfreenect2 #1049: map rgb to depth [Python]

rorpis, 2018-12-22:
Currently I'm working on a project were I need to do image processing in python (using the rgb frame). How could you get the depth value of a pixel from the rgb frame, or map a rgb pixel to a depth pixel using the python wrapper?

rorpis, 2018-12-23:
I am using pylibfreenect2 wrapper

Calvin315, 2018-12-25:
If you haven't found the solution.
Try looking at #1021, pylibfreenect2 has the apply function as well. Sorry, I am not too familiar with it.
[libfreenect2 Apply API](https://r9y9.github.io/pylibfreenect2/stable/api.html#registration-and-geometry)

VJoer, 2019-04-24:
When I go to use apply function，I got the following error：

TypeError: Argument 'rgb' has incorrect type (expected pylibfreenect2.libfreenect2.Frame, got int)

This is part of my code：
`registration = Registration(device.getIrCameraParams(),
                                device.getColorCameraParams())
registration.apply(0, 0, 100., cx, cy)
`
I am also not sure how the dz parameter of this function is determined.
`
void apply | ( | int | dx,
-- | -- | -- | --
  |   | int | dy,
  |   | float | dz,
  |   | float & | cx,
  |   | float & | cy
  | ) |   | const

  | dx | Distorted depth coordinate x (pixel)
  | dy | Distorted depth coordinate y (pixel)
  | dz | Depth value (millimeter)
[out] | cx | Undistorted color coordinate x (normalized)
[out] | cy | Undistorted color coordinate y (normalized)
`