OpenKinect/libfreenect2 #1050: Failed to use USB3.1 to open Kinect V2

MikoyChinese, 2018-12-24:
Overview Description:

Version, Platform, and Hardware Bug Found:
libusb = 1.0.22
ubuntu16.04
USB3.1

1. `git log -1 --oneline`
c203ee2 docs: Remove homebrew/versions from instructions

2. `uname -a`
Linux xxx 4.15.0-42-generic #45~16.04.1-Ubuntu SMP Mon Nov 19 13:02:27 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

3. `lsusb -t`
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 10000M
    |__ Port 1: Dev 14, If 0, Class=Hub, Driver=hub/1p, 5000M
        |__ Port 1: Dev 15, If 0, Class=Vendor Specific Class, Driver=, 5000M
        |__ Port 1: Dev 15, If 1, Class=Vendor Specific Class, Driver=, 5000M
        |__ Port 1: Dev 15, If 2, Class=Audio, Driver=snd-usb-audio, 5000M
        |__ Port 1: Dev 15, If 3, Class=Audio, Driver=snd-usb-audio, 5000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
    |__ Port 1: Dev 11, If 0, Class=Hub, Driver=hub/1p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/10p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/16p, 480M
    |__ Port 5: Dev 32, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 7: Dev 31, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
    |__ Port 7: Dev 31, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M

4. `lspci -nn`
00:00.0 Host bridge [0600]: Intel Corporation Device [8086:3ec2] (rev 07)
00:01.0 PCI bridge [0604]: Intel Corporation Sky Lake PCIe Controller (x16) [8086:1901] (rev 07)
00:01.1 PCI bridge [0604]: Intel Corporation Sky Lake PCIe Controller (x8) [8086:1905] (rev 07)
00:08.0 System peripheral [0880]: Intel Corporation Sky Lake Gaussian Mixture Model [8086:1911]
00:14.0 USB controller [0c03]: Intel Corporation Device [8086:a2af]
00:16.0 Communication controller [0780]: Intel Corporation Device [8086:a2ba]
00:17.0 SATA controller [0106]: Intel Corporation Device [8086:a282]
00:1b.0 PCI bridge [0604]: Intel Corporation Device [8086:a2e7] (rev f0)
00:1b.2 PCI bridge [0604]: Intel Corporation Device [8086:a2e9] (rev f0)
00:1b.4 PCI bridge [0604]: Intel Corporation Device [8086:a2eb] (rev f0)
00:1c.0 PCI bridge [0604]: Intel Corporation Device [8086:a290] (rev f0)
00:1c.2 PCI bridge [0604]: Intel Corporation Device [8086:a292] (rev f0)
00:1c.4 PCI bridge [0604]: Intel Corporation Device [8086:a294] (rev f0)
00:1c.5 PCI bridge [0604]: Intel Corporation Device [8086:a295] (rev f0)
00:1c.6 PCI bridge [0604]: Intel Corporation Device [8086:a296] (rev f0)
00:1c.7 PCI bridge [0604]: Intel Corporation Device [8086:a297] (rev f0)
00:1d.0 PCI bridge [0604]: Intel Corporation Device [8086:a298] (rev f0)
00:1f.0 ISA bridge [0601]: Intel Corporation Device [8086:a2c9]
00:1f.2 Memory controller [0580]: Intel Corporation Device [8086:a2a1]
00:1f.3 Audio device [0403]: Intel Corporation Device [8086:a2f0]
00:1f.4 SMBus [0c05]: Intel Corporation Device [8086:a2a3]
00:1f.6 Ethernet controller [0200]: Intel Corporation Ethernet Connection (2) I219-V [8086:15b8]
01:00.0 VGA compatible controller [0300]: NVIDIA Corporation Device [10de:1b06] (rev a1)
01:00.1 Audio device [0403]: NVIDIA Corporation Device [10de:10ef] (rev a1)
02:00.0 VGA compatible controller [0300]: NVIDIA Corporation Device [10de:1b06] (rev a1)
02:00.1 Audio device [0403]: NVIDIA Corporation Device [10de:10ef] (rev a1)
04:00.0 USB controller [0c03]: ASMedia Technology Inc. Device [1b21:2142]
05:00.0 Non-Volatile memory controller [0108]: Lite-On Technology Corporation Device [14a4:5100] (rev 01)

Steps to Reproduce:

1.  sudo LIBUSB_DEBUG=4 ./Protonect
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./Protonect [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000009] [000052ff] libusb: debug [libusb_init] created default context
[... 745 lines cut ...]
[ 3.476015] [00005301] libusb: debug [handle_events] caught a fish on the event pipe
[ 3.476026] [00005301] libusb: debug [handle_events] someone is closing a device
[ 3.476042] [00005301] libusb: debug [libusb_try_lock_events] someone else is closing a device
[ 3.476050] [000052ff] libusb: debug [usbi_remove_pollfd] remove fd 22
[ 3.476056] [00005301] libusb: debug [libusb_event_handler_active] someone else is closing a device
[ 3.476087] [00005301] libusb: debug [libusb_handle_events_timeout_completed] another thread is doing event handling
[Info] [Freenect2DeviceImpl] closed
[ 3.476134] [00005301] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 3.476150] [00005301] libusb: debug [handle_events] poll fds modified, reallocating
[ 3.476173] [00005301] libusb: debug [handle_events] poll() 2 fds with timeout in 100ms
[ 3.576312] [00005301] libusb: debug [handle_events] poll() returned 0
[ 3.576448] [000052ff] libusb: debug [libusb_exit] 
[ 3.576476] [000052ff] libusb: debug [libusb_exit] destroying default context
[ 3.576493] [000052ff] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 3.576502] [000052ff] libusb: debug [handle_events] poll() 2 fds with timeout in 0ms
[ 3.576514] [000052ff] libusb: debug [handle_events] poll() returned 0
[ 3.576527] [000052ff] libusb: debug [libusb_unref_device] destroy device 3.11
[ 3.576543] [000052ff] libusb: debug [libusb_unref_device] destroy device 3.1
[ 3.576562] [000052ff] libusb: debug [libusb_unref_device] destroy device 2.1
[ 3.576570] [000052ff] libusb: debug [libusb_unref_device] destroy device 1.31
[ 3.576610] [000052ff] libusb: debug [libusb_unref_device] destroy device 1.32
[ 3.576629] [000052ff] libusb: debug [libusb_unref_device] destroy device 1.1
[ 3.576641] [000052ff] libusb: debug [usbi_remove_pollfd] remove fd 6
[ 3.576683] [000052ff] libusb: debug [usbi_remove_pollfd] remove fd 8
[ 3.576756] [00005300] libusb: debug [linux_udev_event_thread_main] udev event thread exiting

The-Raijin, 2018-12-24:
Cancel

On Dec 23, 2018 22:59, "MikoyChinese" <notifications@github.com> wrote:

Overview Description:

Version, Platform, and Hardware Bug Found:
libusb = 1.0.22
ubuntu16.04
USB3.1

   1.

   git log -1 --oneline
   c203ee2
   <https://github.com/OpenKinect/libfreenect2/commit/c203ee2cc8896b086a0b6e9729b0db7d51f13e76>
   docs: Remove homebrew/versions from instructions
   2.

   uname -a
   Linux xxx 4.15.0-42-generic #45
   <https://github.com/OpenKinect/libfreenect2/issues/45>~16.04.1-Ubuntu
   SMP Mon Nov 19 13:02:27 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
   3.

   lsusb -t
   /: Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 10000M
   |__ Port 1: Dev 14, If 0, Class=Hub, Driver=hub/1p, 5000M
   |__ Port 1: Dev 15, If 0, Class=Vendor Specific Class, Driver=, 5000M
   |__ Port 1: Dev 15, If 1, Class=Vendor Specific Class, Driver=, 5000M
   |__ Port 1: Dev 15, If 2, Class=Audio, Driver=snd-usb-audio, 5000M
   |__ Port 1: Dev 15, If 3, Class=Audio, Driver=snd-usb-audio, 5000M
   /: Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
   |__ Port 1: Dev 11, If 0, Class=Hub, Driver=hub/1p, 480M
   /: Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/10p, 5000M
   /: Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/16p, 480M
   |__ Port 5: Dev 32, If 0, Class=Human Interface Device, Driver=usbhid,
   12M
   |__ Port 7: Dev 31, If 0, Class=Human Interface Device, Driver=usbhid,
   1.5M
   |__ Port 7: Dev 31, If 1, Class=Human Interface Device, Driver=usbhid,
   1.5M
   4.

   lspci -nn
   00:00.0 Host bridge [0600]: Intel Corporation Device [8086:3ec2] (rev 07)
   00:01.0 PCI bridge [0604]: Intel Corporation Sky Lake PCIe Controller
   (x16) [8086:1901] (rev 07)
   00:01.1 PCI bridge [0604]: Intel Corporation Sky Lake PCIe Controller
   (x8) [8086:1905] (rev 07)
   00:08.0 System peripheral [0880]: Intel Corporation Sky Lake Gaussian
   Mixture Model [8086:1911]
   00:14.0 USB controller [0c03]: Intel Corporation Device [8086:a2af]
   00:16.0 Communication controller [0780]: Intel Corporation Device
   [8086:a2ba]
   00:17.0 SATA controller [0106]: Intel Corporation Device [8086:a282]
   00:1b.0 PCI bridge [0604]: Intel Corporation Device [8086:a2e7] (rev f0)
   00:1b.2 PCI bridge [0604]: Intel Corporation Device [8086:a2e9] (rev f0)
   00:1b.4 PCI bridge [0604]: Intel Corporation Device [8086:a2eb] (rev f0)
   00:1c.0 PCI bridge [0604]: Intel Corporation Device [8086:a290] (rev f0)
   00:1c.2 PCI bridge [0604]: Intel Corporation Device [8086:a292] (rev f0)
   00:1c.4 PCI bridge [0604]: Intel Corporation Device [8086:a294] (rev f0)
   00:1c.5 PCI bridge [0604]: Intel Corporation Device [8086:a295] (rev f0)
   00:1c.6 PCI bridge [0604]: Intel Corporation Device [8086:a296] (rev f0)
   00:1c.7 PCI bridge [0604]: Intel Corporation Device [8086:a297] (rev f0)
   00:1d.0 PCI bridge [0604]: Intel Corporation Device [8086:a298] (rev f0)
   00:1f.0 ISA bridge [0601]: Intel Corporation Device [8086:a2c9]
   00:1f.2 Memory controller [0580]: Intel Corporation Device [8086:a2a1]
   00:1f.3 Audio device [0403]: Intel Corporation Device [8086:a2f0]
   00:1f.4 SMBus [0c05]: Intel Corporation Device [8086:a2a3]
[... 1275 lines cut ...]
[ 3.475327] [000052ff] libusb: debug [libusb_free_transfer] transfer
0x2c61130
[ 3.475341] [000052ff] libusb: debug [libusb_free_transfer] transfer
0x2c617c0
[ 3.475351] [000052ff] libusb: debug [libusb_free_transfer] transfer
0x2c61900
[ 3.475361] [000052ff] libusb: debug [libusb_free_transfer] transfer
0x2c61a40
[ 3.475370] [000052ff] libusb: debug [libusb_free_transfer] transfer
0x2c61b80
[ 3.475380] [000052ff] libusb: debug [libusb_free_transfer] transfer
0x2c61cc0
[ 3.475388] [000052ff] libusb: debug [libusb_free_transfer] transfer
0x2c61e00
[ 3.475397] [000052ff] libusb: debug [libusb_free_transfer] transfer
0x2c61f40
[ 3.475405] [000052ff] libusb: debug [libusb_free_transfer] transfer
0x2c62080
[ 3.475415] [000052ff] libusb: debug [libusb_free_transfer] transfer
0x2c621c0
[ 3.475425] [000052ff] libusb: debug [libusb_free_transfer] transfer
0x2c62300
[ 3.475437] [000052ff] libusb: debug [libusb_free_transfer] transfer
0x2c62440
[ 3.475446] [000052ff] libusb: debug [l

The-Raijin, 2018-12-24:
Sorry about that. It was obviously sent by mistake.

My apologies.

On Sun, Dec 23, 2018, 23:03 Wantuir Junior <jr.wantuir@gmail.com wrote:

> Cancel
>
> On Dec 23, 2018 22:59, "MikoyChinese" <notifications@github.com> wrote:
>
> Overview Description:
>
> Version, Platform, and Hardware Bug Found:
> libusb = 1.0.22
> ubuntu16.04
> USB3.1
>
>    1.
>
>    git [...]

xlz, 2018-12-24:
04:00.0 USB controller [0c03]: ASMedia Technology Inc. Device [1b21:2142]

This doesn't work. Try a different USB port.