OpenKinect/libfreenect2 #1055: Protonect.exe not working

brandelt4, 2019-01-11:
Hey everyone, 

I've installed the libfreenect2 (with some issues here and there), but finally managed to follow through with all the steps for Windows. To test it, I run Protonect.exe which gives me this output and quits right after with no windows popping out or anything else:

```
C:\libfreenect2-master\build>.\install\bin\Protonect.exe
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: .\install\bin\Protonect.exe [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
```

And that's it. I am fairly new to all of this, so any help would be very appreciated. Thanks!

xlz, 2019-01-11:
Can you run it in visual studio (debug build of couse) and see what error it is?

brandelt4, 2019-01-12:
@xlz Thank you for your willingness to help! Given my lack of expertise in this field, I am struggling to do this through VS15. Would you be able to tell me what files I need to open through VS and debug? Thank you.

brandelt4, 2019-01-12:
This is probably totally wrong, but I opened the following projects:
https://imgur.com/LUeqr25

Then pressed Local Windows Debugger and got this:
https://imgur.com/3wMRcs3

Once again, thanks for your help.

xlz, 2019-01-13:
In the second picture can you show "Call Stack" instead of "Command Window" so we know where it happens.

xlz, 2019-01-13:
How did you build libusb? Can you use this prebuilt binary https://github.com/libusb/libusb/releases/tag/v1.0.22

brandelt4, 2019-01-13:
@xlz Your advice seems to have worked, but I am still not getting an image.. Here is my output:

```
C:\libfreenect2-master\build>.\install\bin\Protonect.exe
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: .\install\bin\Protonect.exe [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 7 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:3 with serial 008894640247
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 3*1048576 ir: 8*64*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 008894640247
device firmware: 4.0.3917.0
[Info] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Info] [DepthPacketStreamParser] 32 packets were lost
[Info] [DepthPacketStreamParser] 32 packets were lost
[Info] [DepthPacketStreamParser] 8 packets were lost
[Info] [DepthPacketStreamParser] 32 packets were lost
[Info] [DepthPacketStreamParser] 32 packets were lost
[Info] [DepthPacketStreamParser] 16 packets were lost
[Info] [DepthPacketStreamParser] 32 packets were lost
[Info] [DepthPacketStreamParser] 32 packets were lost
[Info] [DepthPacketStreamParser] 16 packets were lost
[Info] [DepthPacketStreamParser] 32 packets were lost
[Info] [DepthPacketStreamParser] 32 packets were lost
[Info] [DepthPacketStreamParser] 16 packets were lost
[Info] [DepthPacketStreamParser] 32 packets were lost
[Info] [DepthPacketStreamParser] 32 packets were lost
[Info] [DepthPacketStreamParser] 16 packets were lost
[Info] [DepthPacketStreamParser] 32 packets were lost
[Info] [DepthPacketStreamParser] 32 packets were lost
[Info] [DepthPacketStreamParser] 16 packets were lost
[Info] [DepthPacketStreamParser] 32 packets were lost
[Info] [DepthPacketStreamParser] 32 packets were lost
[Info] [DepthPacketStreamParser] 16 packets were lost
[Info] [DepthPacketStreamParser] 32 packets were lost
[Info] [DepthPacketStreamParser] 32 packets were lost
[Info] [DepthPacketStreamParser] 16 packets were lost
[Info] [DepthPacketStreamParser] 32 packets were lost
[Info] [DepthPacketStreamParser] 32 packets were lost
[Info] [DepthPacketStreamParser] 16 packets were lost
[Info] [DepthPacketStreamParser] 33 packets were lost
[Info] [DepthPacketStreamParser] 33 packets were lost
The viewer is turned off. Received 100 frames. Ctrl-C to stop.
[Info] [DepthPacketStreamParser] 32 packets were lost
[Info] [DepthPacketStreamParser] 33 packets were lost
[Info] [DepthPacketStreamParser] 33 packets were lost
[Info] [DepthPacketStreamParser] 35 packets were lost
[Info] [DepthPacketStreamParser] 32 packets were lost
[Info] [DepthPacketStreamParser] 33 packets were lost
...
```

Thanks again for your help :)

xlz, 2019-01-13:
OpenGL support is not built. Did you have GLFW detected? What's the output of cmake?

brandelt4, 2019-01-13:
```
C:\libfreenect2-master\build>cmake .. -G "Visual Studio 14 2015 Win64"
-- Selecting Windows SDK version  to target Windows 10.0.17134.
-- The C compiler identification is MSVC 19.0.24215.1
-- The CXX compiler identification is MSVC 19.0.24215.1
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio 14.0/VC/bin/x86_amd64/cl.exe
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio 14.0/VC/bin/x86_amd64/cl.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio 14.0/VC/bin/x86_amd64/cl.exe
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio 14.0/VC/bin/x86_amd64/cl.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- using tinythread as threading library
-- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE)
CMake Warning (dev) at CMakeLists.txt:94 (FIND_PACKAGE):
  Policy CMP0074 is not set: find_package uses <PackageName>_ROOT variables.
  Run "cmake --help-policy CMP0074" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  Environment variable LibUSB_ROOT is set to:

    C:\libfreenect2-master\depends\libusb\MS64\dll\libusb-1.0.dll

  For compatibility, CMake is ignoring the variable.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found LibUSB: C:/libfreenect2-master/depends/libusb/MS64/dll/libusb-1.0.lib
-- Could NOT find JPEG (missing: JPEG_LIBRARY JPEG_INCLUDE_DIR)
-- Could NOT find TegraJPEG (missing: TegraJPEG_LIBRARIES TegraJPEG_INCLUDE_DIRS TegraJPEG_L4T_OK TegraJPEG_DRIVER_OK TegraJPEG_WORKS)
CMake Warning (dev) at CMakeLists.txt:243 (FIND_PACKAGE):
  Policy CMP0074 is not set: find_package uses <PackageName>_ROOT variables.
  Run "cmake --help-policy CMP0074" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  Environment variable TurboJPEG_ROOT is set to:

    C:\libjpeg-turbo64\bin\turbojpeg.dll

  For compatibility, CMake is ignoring the variable.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Performing Test TURBOJPEG_WORKS
-- Performing Test TURBOJPEG_WORKS - Success
-- Found TurboJPEG: C:/libjpeg-turbo64/lib/turbojpeg.lib
-- Could NOT find GLFW3 (missing: GLFW3_LIBRARIES GLFW3_INCLUDE_DIRS)
-- Found OpenGL: opengl32
-- Looking for CL_VERSION_2_0
-- Looking for CL_VERSION_2_0 - not found
-- Looking for CL_VERSION_1_2
-- Looking for CL_VERSION_1_2 - not found
-- Looking for CL_VERSION_1_1
-- Looking for CL_VERSION_1_1 - not found
-- Looking for CL_VERSION_1_0
-- Looking for CL_VERSION_1_0 - not found
-- Could NOT find OpenCL (missing: OpenCL_LIBRARY OpenCL_INCLUDE_DIR)
CUDA_TOOLKIT_ROOT_DIR not found or specified
-- Could NOT find CUDA (missing: CUDA_TOOLKIT_ROOT_DIR CUDA_NVCC_EXECUTABLE CUDA_INCLUDE_DIRS CUDA_CUDART_LIBRARY)
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Failed
-- Performing Test COMPILER_HAS_DEPRECATED
-- Performing Test COMPILER_HAS_DEPRECATED - Success
-- Linking with these libraries:
 C:/libfreenect2-master/depends/libusb/MS64/dll/libusb-1.0.lib
 C:/libjpeg-turbo64/lib/turbojpeg.lib
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE)
-- Configurating examples
-- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE)
-- Using in-tree freenect2 target
-- Could NOT find GLFW3 (missing: GLFW3_LIBRARIES GLFW3_INCLUDE_DIRS)
-- Could NOT find OpenNI2 (missing: OpenNI2_LIBRARY_DIR OpenNI2_INCLUDE_DIRS)
-- Feature list:
--   CUDA    no
--   CXX11    disabled
--   Examples    yes
--   OpenCL    no
--   OpenGL    no
--   OpenNI2    no
--   TegraJPEG    no
--   Threading    tinythread
--   TurboJPEG    yes
--   VAAPI    no
--   VideoToolbox    no (Apple only)
--   streamer_recorder    disabled
-- Configuring done
-- Generating done
-- Build files have been written to: C:/libfreenect2-master/build
```

I guess not?

brandelt4, 2019-01-13:
@xlz thanks so much for your help! it finally worked after I reinstalled glfw :)

closed 2019-01-14 by xlz