OpenKinect/libfreenect2 #1065: OpenNI2: can't turn logging off

pierrep, 2019-03-13:
Using latest git and OpenNI 2.2.0.33, I can't turn off the libfreenect2-openni logging. I've tried a bunch of different things, like using the environment variable, trying stuff via the libfreenect api and openni api and nothing seems to work.

Any suggestions?

pierrep, 2019-03-13:
Ok so I've managed to get the environment variable to work using:

`setenv("LIBFREENECT2_LOGGER_LEVEL","none",true);`

This however does nothing:
```
    check_error(openni::OpenNI::setLogMinSeverity(level));
    check_error(openni::OpenNI::setLogConsoleOutput(false));
```
Still curious if there's another way to do this other than via environment variables.

xlz, 2019-03-19:
libfreenect2-openni is part of libfreenect2, so only libfreenect2 can turn it off.

I would suggest setting this env var is the easiest thing to do. Programmatically you need to call `setGlobalLogger()` then libfreenect2 passes logging messages to your own logger.

closed 2019-03-19 by xlz