OpenKinect/libfreenect2 #1066: Undefined reference to*** when `make` [Ubuntu 16.04]

He-Rong, 2019-03-19:
- Platform: Ubuntu 16.04

I installed dependencies accordding to the guide for ubuntu 16.04, and libusb, libopenni2 & OpenGL were installed well.
But met some problem when make libfreenect2 .

result of `make:`
`nvidia@tegra-ubuntu:~/libfreenect2/build$ make
[  5%] Built target generate_resources_tool
[ 69%] Built target freenect2
[ 89%] Built target freenect2-openni2
[ 92%] Linking CXX executable ../bin/Protonect
../lib/libfreenect2.so.0.2.0: undefined reference to 'drmAuthMagic'
../lib/libfreenect2.so.0.2.0: undefined reference to 'drmGetDeviceNameFromFd'
../lib/libfreenect2.so.0.2.0: undefined reference to 'drmFreeVersion'
../lib/libfreenect2.so.0.2.0: undefined reference to 'drmGetVersion'
../lib/libfreenect2.so.0.2.0: undefined reference to 'drmGetClient'
../lib/libfreenect2.so.0.2.0: undefined reference to 'drmGetMagic'                       
collect2: error: ld returned 1 exit status
examples/CMakeFiles/Protonect.dir/build.make:158: recipe for target 'bin/Protonect' failed
make[2]: *** [bin/Protonect] Error 1
CMakeFiles/Makefile2:246: recipe for target 'examples/CMakeFiles/Protonect.dir/all' failed
make[1]: *** [examples/CMakeFiles/Protonect.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2`

Any idea for this?

xlz, 2019-03-19:
```
../lib/libfreenect2.so.0.2.0: undefined reference to 'drmAuthMagic'
```
This means you enabled OpenCL or VAAPI, both of which aren't available on Jetson. So you want to disable them with `cmake -DENABLE_OPENCL=OFF -DENABLE_VAAPI=OFF`.

He-Rong, 2019-03-19:
it works. thanks a lot :)

closed 2019-03-19 by xlz