OpenKinect/libfreenect2 #1071: [Error] [VaapiRgbPacketProcessorImpl] num_entrypoints >= 1 && num_entrypoints <= max_entrypoints failed

RemoteCMeijer, 2019-04-10:
Overview Description:
While running `LIBUSB_DEBUG=3 ./Protonect` I keep running in the error: `[Error] [VaapiRgbPacketProcessorImpl] num_entrypoints >= 1 && num_entrypoints <= max_entrypoints failed` and while the system still works, after a while it crashes. 

1. `git log -1 --`
`fatal: Not a git repository (or any of the parent directories): .git`

2. `uname -a`
`Linux tastbaar-landschap 4.15.0-47-generic #50~16.04.1-Ubuntu SMP Fri Mar 15 16:06:21 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux`

3. `lsusb -t`
```
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/10p, 5000M
    |__ Port 2: Dev 24, If 0, Class=Hub, Driver=hub/1p, 5000M
        |__ Port 1: Dev 25, If 0, Class=Vendor Specific Class, Driver=, 5000M
        |__ Port 1: Dev 25, If 1, Class=Vendor Specific Class, Driver=, 5000M
        |__ Port 1: Dev 25, If 2, Class=Audio, Driver=snd-usb-audio, 5000M
        |__ Port 1: Dev 25, If 3, Class=Audio, Driver=snd-usb-audio, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/16p, 480M
    |__ Port 2: Dev 15, If 0, Class=Hub, Driver=hub/1p, 480M
    |__ Port 6: Dev 2, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
    |__ Port 7: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
    |__ Port 7: Dev 3, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M
    |__ Port 14: Dev 11, If 0, Class=Mass Storage, Driver=usb-storage, `480M`
```

4. `lspci -nn`
```
00:00.0 Host bridge [0600]: Intel Corporation Xeon E3-1200 v6/7th Gen Core Processor Host Bridge/DRAM Registers [8086:591f] (rev 05)
00:01.0 PCI bridge [0604]: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor PCIe Controller (x16) [8086:1901] (rev 05)
00:02.0 VGA compatible controller [0300]: Intel Corporation HD Graphics 630 [8086:5912] (rev 04)
00:14.0 USB controller [0c03]: Intel Corporation 200 Series/Z370 Chipset Family USB 3.0 xHCI Controller [8086:a2af]
00:14.2 Signal processing controller [1180]: Intel Corporation 200 Series PCH Thermal Subsystem [8086:a2b1]
00:16.0 Communication controller [0780]: Intel Corporation 200 Series PCH CSME HECI #1 [8086:a2ba]
00:16.3 Serial controller [0700]: Intel Corporation Device [8086:a2bd]
00:17.0 SATA controller [0106]: Intel Corporation 200 Series PCH SATA controller [AHCI mode] [8086:a282]
00:1f.0 ISA bridge [0601]: Intel Corporation 200 Series PCH LPC Controller (Q270) [8086:a2c6]
00:1f.2 Memory controller [0580]: Intel Corporation 200 Series/Z370 Chipset Family Power Management Controller [8086:a2a1]
00:1f.3 Audio device [0403]: Intel Corporation 200 Series PCH HD Audio [8086:a2f0]
00:1f.4 SMBus [0c05]: Intel Corporation 200 Series/Z370 Chipset Family SMBus Controller [8086:a2a3]
00:1f.6 Ethernet controller [0200]: Intel Corporation Ethernet Connection (5) I219-LM [8086:15e3]
01:00.0 VGA compatible controller [0300]: NVIDIA Corporation GP104 [GeForce GTX 1070] [10de:1b81] (rev a1)
01:00.1 Audio device [0403]: NVIDIA Corporation GP104 High Definition Audio Controller [10de:10f0] (rev a1)

```
Steps to Reproduce:

1. install libfreenect2
2. run LIBUSB_DEBUG=3 ./Protonect

Actual Results:
```
LIBUSB_DEBUG=3 /home/tastbaar/tangiblelandscape2/libfreenect2-0.2.0/build/bin/Protonect 
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: /home/tastbaar/tangiblelandscape2/libfreenect2-0.2.0/build/bin/Protonect [-gpu=<id>] [gl | cl | cuda | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 8 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:27 with serial 022615543847
[Info] [Freenect2Impl] found 1 devices
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/gallium_drv_video.so
libva info: Found init function __vaDriverInit_0_39
libva info: va_openDriver() returns 0
[Info] [VaapiRgbPacketProcessorImpl] driver: mesa gallium vaapi
[Error] [VaapiRgbPacketProcessorImpl] num_entrypoints >= 1 && num_entrypoints <= max_entrypoints failed
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9729
[Debug] [Freenect2DeviceImpl] status 0x090000: 9731
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 022615543847
device firmware: 4.0.3917.0
[Debug] [DepthPacketStreamParser] not all subsequences received 0
[Debug] [DepthPacketStreamParser] not all subsequences received 512
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Info] [OpenGLDepthPacketProcessor] avg. time: 8.56544ms -> ~116.748Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 12.7021ms -> ~78.7274Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 8.49012ms -> ~117.784Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 12.3307ms -> ~81.0984Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 8.41178ms -> ~118.881Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 12.4345ms -> ~80.4215Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 8.46718ms -> ~118.103Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 12.2822ms -> ~81.4189Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 8.52103ms -> ~117.357Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 12.399ms -> ~80.6516Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 8.45561ms -> ~118.265Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 12.2425ms -> ~81.6826Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 8.47932ms -> ~117.934Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 12.3943ms -> ~80.6822Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 8.43529ms -> ~118.55Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 12.3478ms -> ~80.9863Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 8.40248ms -> ~119.013Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 12.3449ms -> ~81.0053Hz

```

Expected Results:
Working system, which keeps running and using ./Protonect

Actually the systems keeps crashing on me while running without giving an error.
The error: 
`[Error] [VaapiRgbPacketProcessorImpl] num_entrypoints >= 1 && num_entrypoints <= max_entrypoints failed`

Is this a known error? Do you have any hints on how to approach this?

If more logs are needed, let me know.
Thanks, 
Marc

xlz, 2019-04-19:
VAAPI may be unstable for you. You can disable it with cmake -DENABLE_VAAPI=OFF.

RemoteCMeijer, 2019-04-23:
Thank you for your response! 
It have ran the recommended command in
`/home/tastbaar/tangiblelandscape2/libfreenect2-0.2.0/`

Gives the following output:
```
cmake -DENABLE_VAAPI=OFF
-- using tinythread as threading library
-- Could NOT find TegraJPEG (missing:  TegraJPEG_LIBRARIES TegraJPEG_INCLUDE_DIRS TegraJPEG_L4T_OK TegraJPEG_DRIVER_OK TegraJPEG_WORKS) 
CUDA_TOOLKIT_ROOT_DIR not found or specified
-- Could NOT find CUDA (missing:  CUDA_TOOLKIT_ROOT_DIR CUDA_NVCC_EXECUTABLE CUDA_INCLUDE_DIRS CUDA_CUDART_LIBRARY) 
-- Linking with these libraries: 
 /usr/lib/x86_64-linux-gnu/libusb-1.0.so
 pthread
 /usr/lib/x86_64-linux-gnu/libturbojpeg.so.0
 /usr/lib/x86_64-linux-gnu/libglfw.so
 /usr/lib/x86_64-linux-gnu/libGL.so
 /usr/lib/x86_64-linux-gnu/libOpenCL.so
-- Could NOT find Doxygen (missing:  DOXYGEN_EXECUTABLE) 
-- Configurating examples
-- Using in-tree freenect2 target
-- Feature list:
--   CUDA    no
--   CXX11    disabled
--   Examples    yes
--   OpenCL    yes
--   OpenGL    yes
--   OpenNI2    yes
--   TegraJPEG    no
--   Threading    tinythread
--   TurboJPEG    yes
--   VAAPI    disabled
--   VideoToolbox    no (Apple only)
-- Configuring done
-- Generating done
-- Build files have been written to: /home/tastbaar/tangiblelandscape2/libfreenect2-0.2.0
```

Running LINUSB_DEBUG=3 ./Protonect:
```
LIBUSB_DEBUG=3 /home/tastbaar/tangiblelandscape2/libfreenect2-0.2.0/build/bin/Protonect Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: /home/tastbaar/tangiblelandscape2/libfreenect2-0.2.0/build/bin/Protonect [-gpu=<id>] [gl | cl | cuda | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 8 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:9 with serial 022615543847
[Info] [Freenect2Impl] found 1 devices
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/gallium_drv_video.so
libva info: Found init function __vaDriverInit_0_39
libva info: va_openDriver() returns 0
[Info] [VaapiRgbPacketProcessorImpl] driver: mesa gallium vaapi
[Error] [VaapiRgbPacketProcessorImpl] num_entrypoints >= 1 && num_entrypoints <= max_entrypoints failed
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9729
[Debug] [Freenect2DeviceImpl] status 0x090000: 9731
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 022615543847
device firmware: 4.0.3917.0
[Debug] [DepthPacketStreamParser] not all subsequences received 0
[Info] [OpenGLDepthPacketProcessor] avg. time: 8.35241ms -> ~119.726Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 13.0682ms -> ~76.5216Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 8.16258ms -> ~122.51Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 12.6725ms -> ~78.9109Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 8.14508ms -> ~122.774Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 12.6568ms -> ~79.0087Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 8.12865ms -> ~123.022Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 12.6379ms -> ~79.1269Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 8.17085ms -> ~122.386Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 12.6647ms -> ~78.9598Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 8.12685ms -> ~123.049Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 12.6519ms -> ~79.0397Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 8.31042ms -> ~120.331Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 12.8681ms -> ~77.7117Hz
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] already closed, doing nothing
```

Still produces the same error:
`[Error] [VaapiRgbPacketProcessorImpl] num_entrypoints >= 1 && num_entrypoints <= max_entrypoints failed`

Have a disabled it correctly? The scanner works well for a varying amount of time. Sometimes it functions for 5 minutes, sometimes for 1 minute. Unplugging and replugging the USB always 'solves' the problem.

xlz, 2019-04-23:
I meant for you to clean and rebuild libfreenect2 with cmake and the above options, then VAAPI should be disabled completely. But you seemed to be describing two problems but haven't actually described the second problem, how it crashed.

RemoteCMeijer, 2019-04-23:
Correct me if I am wrong but I have tried to follow your advice. 
It crashes without giving a warning, for example:

```
/home/tastbaar/test_lib/libfreenect2/build/bin/Protonect 
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: /home/tastbaar/test_lib/libfreenect2/build/bin/Protonect [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 8 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:17 with serial 022615543847
[Info] [Freenect2Impl] found 1 devices
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/gallium_drv_video.so
libva info: Found init function __vaDriverInit_0_39
libva info: va_openDriver() returns 0
[Info] [VaapiRgbPacketProcessorImpl] driver: mesa gallium vaapi
[Error] [VaapiRgbPacketProcessorImpl] num_entrypoints >= 1 && num_entrypoints <= max_entrypoints failed
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9729
[Debug] [Freenect2DeviceImpl] status 0x090000: 9731
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 022615543847
device firmware: 4.0.3917.0
[Debug] [DepthPacketStreamParser] not all subsequences received 0
[Info] [OpenGLDepthPacketProcessor] avg. time: 8.20338ms -> ~121.901Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 12.3384ms -> ~81.0477Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 8.19008ms -> ~122.099Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 12.3767ms -> ~80.7973Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 8.68123ms -> ~115.191Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 12.4951ms -> ~80.0315Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 8.46175ms -> ~118.179Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 12.3987ms -> ~80.6537Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 8.15178ms -> ~122.673Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 12.4729ms -> ~80.174Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 8.26236ms -> ~121.031Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 12.2866ms -> ~81.3894Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 8.19575ms -> ~122.014Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 12.2284ms -> ~81.7769Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 8.26432ms -> ~121.002Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 12.3716ms -> ~80.8304Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 8.54234ms -> ~117.064Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 12.4753ms -> ~80.1582Hz
```
end of a normal run, resulting in a crash (might be hardware?)

I've made a new folder for testing
then ran:
```
git clone https://github.com/OpenKinect/libfreenect2.git
cd libfreenect2
```
Followed by: 
```
mkdir build && cd build
cmake .. -DCMAKE_INSTALL_PREFIX=$HOME/freenect2
make
make install
```
And then: 
```
cmake -DENABLE_VAAPI=OFF
```
to disable VAAPI