OpenKinect/libfreenect2 #1075: Streamer recorder build error with OpenCV 4.1

anuppari, 2019-07-18:
## Overview Description:
Missing macros when compiling with OpenCV 4.1

## Version, Platform, and Hardware Bug Found:

1.dfd4eaf (HEAD -> master, origin/master, origin/HEAD) Add android build script and instructions (#1060)
2. Ubuntu 18.04 (Linux s1039351 4.15.0-54-generic #58-Ubuntu SMP Mon Jun 24 10:55:24 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux)

## Steps to Reproduce:

1. Compile with OpenCV 4.1

## Actual Results:

Build errors:
```bash
libfreenect2/tools/streamer_recorder/streamer.cpp: In member function ‘void Streamer::initialize()’:
libfreenect2/tools/streamer_recorder/streamer.cpp:39:32: error: ‘CV_IMWRITE_JPEG_QUALITY’ was not declared in this scope
   compression_params.push_back(CV_IMWRITE_JPEG_QUALITY);
                                ^~~~~~~~~~~~~~~~~~~~~~~
libfreenect2/tools/streamer_recorder/recorder.cpp: In member function ‘void Recorder::initialize()’:
libfreenect2/tools/streamer_recorder/recorder.cpp:81:28: error: ‘CV_IMWRITE_JPEG_QUALITY’ was not declared in this scope
   img_comp_param.push_back(CV_IMWRITE_JPEG_QUALITY); //specify the compression technique
```

## Expected Results:
Successful compilation

AaboutL, 2019-08-14:
Do you fixed this bug?

livemixlove, 2020-04-18:
You can change it to: cv::IMWRITE_JPEG_QUALITY

smokhov, 2020-11-06:
Yeah, OpenCV 4 is not supported... Not yet at least. I am not sure how we can elegantly replace `CV_IMWRITE_JPEG_QUALITY` with `cv::IMWRITE_JPEG_QUALITY` for OpenCV4+ and not when using 3-...

smokhov, 2022-01-01:
Referencing PR #1163 -- hopefully both can be supported somehow.