OpenKinect/libfreenect2 #1080: What is the libfreenect2 license?

fishcu, 2019-08-09:
See title.

I have found two seemingly license-related files in the root of the repository, APACHE20 and GPL2.
However I could not find which one of these licenses applies to the code base of libfreenect2 as a whole. Please clarify!

Can the license be chosen at will? Or is there some other mechanism how the licensing works?

I have found a similar description for the libfreenect2-openni2 dependency:
```
This code is licensed to you under the terms of the Apache License, version
2.0, or, at your option, the terms of the GNU General Public License,
version 2.0. See the APACHE20 and GPL2 files for the text of the licenses,
or the following URLs:
[...]
```

fishcu, 2019-08-13:
Bump.

@floe Do you happen to know?

fishcu, 2019-08-13:
This website mentions license as "other (open)": https://zenodo.org/record/50641

floe, 2019-08-13:
Disclaimer: IANAL, but as far as I understand, you are free to choose which of both licenses to apply to your derivative works.

fishcu, 2019-08-14:
Thank you for the confirmation!

closed 2019-08-14 by fishcu