OpenKinect/libfreenect2 #1081: Using libfreenect2 for getting object position

TomasMerva, 2019-08-11:
Hi, I just want to ensure myself that this is the right library for my application. I want to get [x,y,z] coordinates of any object within Linux operation system (Ubuntu 16). The idea is that I will use OpenCV to recognize object and after that I ll get the depth (z coordinate) of an object from libfreenect2 library. Is this possible? I would like to avoid another libraries, because they are kind of outdated regards to OpenCV, Ubuntu and ROS version.

floe, 2019-08-12:
Yes, that would be relatively straightforward. You can just do object detection with OpenCV on the RGB image, and then get the corresponding z coordinate for each pixel from the depth image using the `Registration` class.

TomasMerva, 2019-08-12:
@floe If I can ask you, do I need to calibrate Kinect somehow? I mean it in terms of getting the correct information from Kinect.

floe, 2019-08-13:
Usually not, the Kinect has an internal factory calibration that's sufficiently accurate for most cases and libfreenect uses that. Only if you need really high precision, then you would need to do a separate calibration

fred-dev, 2019-08-13:
On this topic (sorry to hijack) for some libraries that require a calibration file like aruco marker detection what is best practice, to create a blank calibration file or to calibrate again (it seems then the system will run through double calibration matrices).

TomasMerva, 2019-08-13:
Thank you all for your help :)

closed 2019-08-13 by TomasMerva