OpenKinect/libfreenect2 #1083: Does getIrCameraParams refer to intrinsics before or after rectification?

fishcu, 2019-08-14:
See title.

I would like to use the data as it comes out after applying the rectification (undistortion and color to depth registration). Part of the data is the (pinhole) calibration.
Since I don't know what kind of corrections the rectification does internally, I'm wondering if the parameters for fx, fy, cx, cy I get from `getIrCameraParams` correspond to values I can directly use with rectified data, or if only corresponds to the data before rectification.

Any help is appreciated!

floe, 2019-08-14:
No, these are the parameters that are used _for_ rectification, so they are mostly meaningless for the final aligned/rectified images.

fishcu, 2019-08-14:
Makes sense. Then, the easiest way to get pinhole parameters for the rectified data is to do a calibration with a standard target like a checkerboard, correct?

floe, 2019-08-14:
Now that I think about it, the rectification does primarily use the radial distortion parameters, and fx,fy,cx,cy should still be valid for the rectified image. But if you're doing anything involving noticeable precision requirements, then you're very likely better off with your own calibration anyway.