OpenKinect/libfreenect2 #1087: [bug] bilateral filter = false, edge aware filter = true gives invalid depth

fishcu, 2019-08-21:
Consider the following snippet of code:

```
libfreenect2::Freenect2Device::Config config;
  config.MinDepth = 0.5f;
  config.MaxDepth = 4.5f;
  config.EnableBilateralFilter = false;
  config.EnableEdgeAwareFilter = true;
  device_handle->setConfiguration(config);
```

This will produce an invalid depth output. I have attached an example of such an output (after rgb->depth registration). Note that the float image has been converted to RGB via Matlab's "jet" color map.
[image]

I have tried the following other combinations:
- bilateral = true, edge aware = true
- bilateral = true, edge aware = false
- bilateral = false, edge aware = false

And they all work.

Thus I suspect this is a bug in the driver, possibly caused by the edge aware filter using some temporary output buffer of the bilateral filter even if it is valid?

fishcu, 2019-08-21:
Here's an example with both filters enabled, to show that the rest of the code works:

[image]

xlz, 2019-09-27:
> the edge aware filter using some temporary output buffer of the bilateral filter

It is the case.

Duplicate of #163.

closed 2019-09-27 by xlz