OpenKinect/libfreenect2 #1096: ./bin/Protonect is not working but sudo ./bin/Protonect is

ravijo, 2019-11-23:
I am facing a strange issue. `./bin/Protonect` is not working but `sudo ./bin/Protonect` is working. Is it related to permissions?

Below is the output reported at the terminal-
```
ravi@lab:~/tools/libfreenect2/build/bin$ ./Protonect 
./Protonect: /usr/local/cuda/lib64/libOpenCL.so.1: no version information available (required by /home/ravi/tools/libfreenect2/build/lib/libfreenect2.so.0.2)
./Protonect: /usr/local/cuda/lib64/libOpenCL.so.1: no version information available (required by /home/ravi/tools/libfreenect2/build/lib/libfreenect2.so.0.2)
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./Protonect [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 18 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:3 with serial 510760242542
[Info] [Freenect2Impl] found 1 devices
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns -1
libva error: va_getDriverName() failed with unknown libva error,driver_name=(null)
[Error] [VaapiRgbPacketProcessorImpl] vaInitialize(display, &major_ver, &minor_ver): unknown libva error
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 1073750044
[Debug] [Freenect2DeviceImpl] status 0x090000: timeout
[Debug] [Freenect2DeviceImpl] status 0x090000: 1073750044
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 510760242542
device firmware: 4.0.3916.0
timeout!
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
```

Below is the kernel infomation-
```
ravi@lab:~/tools/libfreenect2/build/bin$ uname -a
Linux lab 4.4.0-148-generic #174~14.04.1-Ubuntu SMP Thu May 9 08:17:37 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
```

Below is the information about GCC-
```
ravi@lab:~/tools/libfreenect2/build/bin$ gcc --version
gcc (Ubuntu 5.5.0-12ubuntu1~14.04) 5.5.0 20171010
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Below is the information about OS-
```
ravi@lab:~/tools/libfreenect2/build/bin$ cat /etc/issue
Ubuntu 14.04.6 LTS \n \l
```
Below is the information about the graphics card-
```
ravi@lab:~/tools/libfreenect2/build/bin$ nvidia-smi 
Sat Nov 23 19:57:28 2019       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 384.130                Driver Version: 384.130                   |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  GeForce GTX 1080    Off  | 00000000:01:00.0  On |                  N/A |
| 46%   52C    P8    17W / 180W |    474MiB /  8105MiB |      3%      Default |
+-------------------------------+----------------------+----------------------+
```

Any workaround, please?

floe, 2019-11-25:
Quoting from the README: "Set up udev rules for device access: `sudo cp ../platform/linux/udev/90-kinect2.rules /etc/udev/rules.d/`, then replug the Kinect." Give this a try, should fix it.

ravijo, 2019-11-26:
Thank you very much. Allow me to explain the complete story. It was not the fresh install. Since [iai_kinect2](https://github.com/code-iai/iai_kinect2) was showing errors, I rebuild `libfreenect2`.

> Quoting from the README: "Set up udev rules for device access: `sudo cp ../platform/linux/udev/90-kinect2.rules /etc/udev/rules.d/`, then replug the Kinect." Give this a try, should fix it.

Surprisingly, it is working now after copying these rules.

Thanks again.

closed 2019-11-26 by ravijo