OpenKinect/libfreenect2 #1099: Using Kinect v2 RGB camera as a webcam

sonatbaltaci, 2019-12-16:
Hello,

Is it possible to use Kinect v2 as a webcam in Ubuntu to access from other applications such as Cheese? In other words, I would like to register the RGB camera of Kinect v2 as /dev/videoX.
I have tried:

https://github.com/yoshimoto/gspca-kinect2

as a driver. However, it didn't work with the test program Protonect.

Thank you in advance.

floe, 2020-01-13:
You can either use the gspca-kinect2 driver, _or_ Protonect. They will not work together at the same time.

SterlingButters, 2020-04-14:
@floe but the `libfreenect2` driver is not `V4L2` and thus will not be detected by `.getUserMedia` in the browser for a webcam correct? - at least I have not had such luck

floe, 2020-04-14:
Correct. You will need gspca-kinect2 for that.

closed 2020-04-14 by floe

gytisrepecka, 2020-12-25:
Is it ok to install both `Protonect` (and it's prerequisites) and `gspca-kinect2` on the same system?

SterlingButters, 2020-12-26:
I have not had an issue unless you try to run them at the same time. Also for future comers running ubuntu: https://github.com/grandchild/gspca-kinect2/pull/1