OpenKinect/libfreenect2 #1100: Get Raw depth data in python

mattiaottoborgo, 2019-12-27:
hi, i was tinkering with the kinect sensor with python and opencv.
Is there a way to get the distance from the sensor in millimeters? I ask because i don't find any examples or documentation of it in python, thanks

floe, 2020-01-13:
The data in the depth image is in mm. See here: https://openkinect.github.io/libfreenect2/classlibfreenect2_1_1Frame.html#a1d1cfd8ffb84e947f82999c682b666a7

tavishm, 2020-04-17:
You can try using [freenect2](https://rjw57.github.io/freenect2-python/), it worked for me on ubuntu 14.04 and raspberry pi 4 with python. It's pretty simple to use and gives color and depth pics. It's based on libfreenect2. It has good documentation.