OpenKinect/libfreenect2 #1102: Failed to connect to display

itsyoshio, 2020-01-05:
Overview Description:
I can't get Video from Protonect, or rather i cannot display it on my Screen.
Haven't had any luck with either X11 or Wayland. (tried Deepin and Gnome Wayland / Gnome X11).
Any pointers in the right direction are greatly appreciated.

Version, Platform, and Hardware Bug Found:

1. `da68a71 (HEAD -> master, origin/master, origin/HEAD) Update TegraJPEG detection`
2. `Linux yoshio-notebook 5.3.18-1-surface #1 SMP PREEMPT Sat, 28 Dec 2019 22:53:29 +0000 x86_64 GNU/Linux`
3. 
```
Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/6p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/3p, 5000M
        |__ Port 2: Dev 3, If 0, Class=Hub, Driver=hub/1p, 5000M
            |__ Port 1: Dev 4, If 0, Class=Vendor Specific Class, Driver=, 5000M
            |__ Port 1: Dev 4, If 1, Class=Vendor Specific Class, Driver=, 5000M
            |__ Port 1: Dev 4, If 2, Class=Audio, Driver=snd-usb-audio, 5000M
            |__ Port 1: Dev 4, If 3, Class=Audio, Driver=snd-usb-audio, 5000M

```
4. `00:14.0 USB controller [0c03]: Intel Corporation Sunrise Point-LP USB 3.0 xHCI Controller [8086:9d2f] (rev 21)`

Steps to Reproduce:

1. Build current libfreenect2 from Github
2. run /build/Protonect

Actual Results:
`$ ./Protonect`
```
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./Protonect [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 9 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:4 with serial 298089633947
[Info] [Freenect2Impl] found 1 devices
[Error] [OpenGLDepthPacketProcessorImpl] GLFW error 65544 Wayland: Failed to connect to display
[Error] [OpenGLDepthPacketProcessor] Failed to initialize GLFW.
```

Expected Results:
- Display Sensor Results

Reproducibility:
- Unknown

Additional Information:
- Kinect v2 with Official Microsoft Kinect One to Windows Adapter.
- Bought a Second Pair (unofficial [Kinect One Windows Adapter](https://www.amazon.de/dp/B07GFGBF6Q/ref=sr_1_3?keywords=Kinect+v2&qid=1578252389&sr=8-3) + Kinect v2) where i experience the same result.
- `./Protonect cpu` results in `$ GLFW error 65537 The GLFW library is not initialized` being displayed 5 times.

floe, 2020-01-13:
Can you run `glxgears`, does that work? And have you tried installing `libglfw3-wayland` (just a guess, I'm not a Wayland user myself).

itsyoshio, 2020-01-17:
Hi floe,

Yes i can run glxgears without err.
I'm on Arch so the corresponding Package would be glfw-wayland, which is installed.

Thank you so far.

floe, 2020-01-20:
Maybe related: #946

Beyley, 2021-05-09:
i seem to be getting this error too, even though i dont have wayland installed at all

JustCryen, 2024-02-21:
I have issues getting it to work as well.
There are two `glfw` packages on Arch: `glfw-wayland` and `glfw-x11`. Both conflicting with each other. For some reason libfreenect2 defaults to `glfw-wayland`. Does it matter if I'm trying to use libfreenect2 on x11?

JustCryen, 2024-03-02:
Yeah, sorry, I forgot to mention that I got it to launch `Protonect` by changing to glfw-x11 but not without further problems.
For now I'm still having issues with VAAPI on an AMD card, `Protonect` works only if I use `export LIBVA_DRIVER_NAME=i965` as currently `libfreenect2` doesn't seem to support VAAPI on AMD…
The current "solution" seems to be to remove libva altogether but that's not really an option, is it?
Btw, I've reverted the biro-mark's fix, it looks like it isn't really needed.