OpenKinect/libfreenect2 #1103: Usb to ethernet raw data repeater and raw data reprocessing.

alec-g, 2020-01-11:
Hello, I am looking at a way to extend the working distance of traditional usb 3.0, currently for my project i am running 4 kinects on one system, and using usb active extensions to extend the range of the kinects. I am having intermittent issues with the usb active extensions, particulary when i start to daisy chain, the kinects are either not recognized at all, are recognized but don't open correctly, or open and then crash shortly after. However i can run the four fairly stable if i don't daisy chain. If anyone has any ideas on making this behave better im all ears.

My team's preferred solution originally was to run the kinect's using four seperate computers, some single board computer like the new raspberry pi 4 would be nice, but as i understand, these will not meet performance standards for decent depth processing frame rate (we need 30hz +).

So the main question, is it possible to repeat the raw data of the kinects over ethernet? My first idea would be to use the DumpPacketPipeline, which constructs a Frame instance with the raw packet data, and then send this over ethernet (perhaps using ROS). This data would go into a high-end pc to perform processing on the RGB and Depth packets, how could i re-input this raw data (received over etherner) into libfreenect2 to perform rgb and depth packet processing using appropriate methods (eg. OpenCL and Cuda) to then spit out the processed rgb and depth images?

Has anyone tried doing something similar before? Any thoughts / ideas is much appreciated.

floe, 2020-01-12:
OK, so this is somewhat non-trivial. The raw data of the Kinect v2 pretty much maxes out the available USB3 bandwidth of a single bus controller (~ 5 GBit/s), that's the initial reason why even minor cabling issues can cause noticeable problems. 

While your network tunneling idea might work in principle, this also means that you will need a network with 10 GBit Ethernet connections, and your repeater nodes need to shovel 5 GBit per second over the network, while the receiving machine needs to reliably download 20 GBit/s.

If your installation really needs to be spread out so that the original cables are not long enough, then I would probably rather pick something like a low-end NUC with Core i3 and use four of those for depth processing, and then just send the depth/RGB streams over the network. I

floe, 2020-01-12:
FWIW, I just checked and the NUC I used for depth processing was bought in 2015, so any low-end Core i3 with Intel HD GPU shouldn't have any issues with live-streaming a single Kinect v2.

fred-dev, 2020-01-13:
IMHO you will be better off changing your cabling for a system that is designed to do the distance you want, daisy chaining extenders is very problematic, they are usually designed to cover their own distance, the streaming option will take a lot of work and high bandwidth network to make it work like a direct attached unit. Can you verify that the cameras all work fine without the extenders? If so there are some cheap and up to 50m optical extension cables from some Chinese sourced supply sites that will work quite well. If you want to send full depth packets you will need to do some compression and reconstruct on the other end. If you can sacrifice some depth precision, encoding the depth into the 8 bit alpha channel of an RGBA video stream and using a transport stream like NDI to send a compressed signal can get you a pretty good low overhead remote system.

alec-g, 2020-01-15:
Thanks all for the prompt replies. @floe i have looked into the NUC option and done some research on other possible small scale computers, however in Australia sourcing an appropriate low-end computer that could do the job is around $300 - $350 AUD, multiply this by four and it get's pretty expensive, unfortunately we are trying to keep the cost down for this job.

@fred-dev yes the four cameras work on one system without the extenders, i am also able to get all four camera's working on one system with the extenders if i **dont** daisy chain. I like the idea of using fibre and will have a further look into it.

On another note, back to the usb extensions, recently i purchased a 10m USB 3.0 extension (previously have been using 2 x 5m extensions and daisy chaining), however i am getting a strange issue whilst using it. Currently i am running a single high-end PC, with 3 x Pci USB 3.0 expansion boards. I run 1 x kinect off the main motherboard USB3.0 hub and the the other 3 off an individual expansion. 

The problem is, the new 10M usb extension only works off the motherboard USB 3.0 hub, when plugged into one of the PCI expansion boards, i am able to get RGB data, but no depth / IR data is coming through. Anyone have any thoughts as to why this might be? Using the old 5M extensions it was working fine on these hubs

fred-dev, 2020-01-15:
Maybe the cards have less wattage, or a different chip that rectified the modulation differently, copper cables are not good for USB 3 that long, I wouldn't bother to troubleshoot, (but maybe your pcie cards also have a 4 pin power input) go for the fibre extensions, they are probably about the same price as a 10 m copper. Also there are some pcie USB cards that have 4 independent USB 3.0 controllers on them and they use a pciex4 slot, much better than losing 3 slots for USB cards and sending some through the motherboard.

floe, 2020-01-15:
I agree, USB3 was and still is a clusterf*ck. Whether any given combination of host, cabling and device works is just random luck, e.g. there are some older host controllers which just don't work at all with the Kinect v2.

BTW, in any scenario where you're only getting RGB data, that means that USB bulk transfers work, but USB isochronous transfers (which have stricter timing requirements) don't.

xlz, 2020-01-27:
I agree optical fiber cables are viable. I used 20m optical cables (https://www.fibbrtech.com/) with USB3 Vision cameras at full 5Gb/s data rate. It mostly worked fine but some other brands were not and they seemed finicky about connection setup. Be sure to test and verify both RGB and depth data.

High-end Ethernet runs on optical too, so USB over Ethernet isn't a whole lot different from USB over optical.