OpenKinect/libfreenect2 #1110: Build: target "Threads::Threads" was not found.

oneandonlyoddo, 2020-02-20:
I am trying to build libfreenect2 on Windows 10 with Visual Studio 2019 Community Edition.
Configuring runs without problems.
Then there are lots of CMake warnings like this one:
```
CMake Warning (dev) at examples/CMakeLists.txt:76 (ADD_EXECUTABLE):
Policy CMP0028 is not set: Double colon in target name means ALIAS or
IMPORTED target.  Run "cmake --help-policy CMP0028" for policy details.
Use the cmake_policy command to set the policy and suppress this warning.

Target "Protonect" links to target "Threads::Threads" but the target was
not found.  Perhaps a find_package() call is missing for an IMPORTED
target, or an ALIAS target is missing?
This warning is for project developers.  Use -Wno-dev to suppress it.
```

As far as I can tell these warnings come up for every target that lists _Threads::Threads.lib_ in _AdditionalDependencies_.

Then it starts building but fails at some point due to a link error with Threads:
```
  -- Generating done
  -- Build files have been written to: C:/libfreenect2/build
  transfer_pool.cpp
  event_loop.cpp
  usb_control.cpp
  allocator.cpp
  frame_listener_impl.cpp
  packet_pipeline.cpp
  rgb_packet_stream_parser.cpp
  rgb_packet_processor.cpp
  depth_packet_stream_parser.cpp
  depth_packet_processor.cpp
  cpu_depth_packet_processor.cpp
  resource.cpp
  command_transaction.cpp
  registration.cpp
  logging.cpp
  libfreenect2.cpp
  tinythread.cpp
  turbo_jpeg_rgb_packet_processor.cpp
  flextGL.cpp
  opengl_depth_packet_processor.cpp
  Generating Code...
  Compiling...
  opencl_depth_packet_processor.cpp
  opencl_kde_depth_packet_processor.cpp
  Generating Code...
LINK : fatal error LNK1104: cannot open file 'Threads::Threads.lib' [C:\libfreenect2\build\freenect2.vcxproj]
```
_Threads::Threads.lib_ doesn't produce any good Google results for me so I don't understand if I am missing something third party or if it is part of libfreenect2. 
Any ideas on how I can fix this link?

oneandonlyoddo, 2020-02-21:
I know opened the solution in VS 15 and removed _Threads::Threads.lib_ from the additional dependencies for _freenect2_, _freenect2-openni2_ and _Protonect_.
Managed to build and Protonect runs. I'll see if I run into any issues.

Andy-Cheng, 2020-06-09:
I encountered the same problem on VS2019.

NicholasNelsonwood, 2020-06-15:
Try adding find_package(Threads) to the CMake file, that seems to have worked for me - must be something about when the library is added vs when it is called?

Andy-Cheng, 2020-06-20:
> Try adding find_package(Threads) to the CMake file, that seems to have worked for me - must be something about when the library is added vs when it is called?

It works! Thanks for saving my time ^^

BTW, just as @NicholasNelsonwood  mentioned, I added FIND_PACKAGE(Threads) before INCLUDE(SetupLibfreenect2Threading) in CMakeList.txt.