OpenKinect/libfreenect2 #1112: Multiple read access to one Kinect v2

mnemllr, 2020-02-28:
Overview Description:

I want to use one Kinect v2 camera for multiple applications on one computer. If I try to to get access with a second application using libfreenect2, both applications crash. 
Is it somehow possbile to do that? 

I am using Ubuntu 16.04.

Thank you.

floe, 2020-02-28:
I guess the easiest way to do that would be to use ROS as middleware, and https://github.com/code-iai/iai_kinect2 as Kinect v2 driver.

mnemllr, 2020-02-28:
Thanks for your answer.
So with libfreenect2 it is not possible, right?
Because one of the apps is an open source software which is using libfreenect2. I do not want to change their code base.

floe, 2020-03-07:
Correct, libfreenect2 assumes exclusive hardware access, so two separate instances will just crash.

closed 2020-03-07 by floe