OpenKinect/libfreenect2 #1115: Protonect does not detect Kinect on Raspberry Pi 4

peterjohnzking, 2020-03-26:
I followed the instructions to install libfreenect2 on linux and tried to run protonect. It said that the device was not found:

pi@raspberrypi:~/libfreenect2/build/bin $ ./Protonect -noviewer cpu
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./Protonect [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 9 usb devices connected
[Info] [Freenect2Impl] found 0 devices
no device connected!

I had also tried to run it without the flags. 

I ensured that the file 90-kinect2.rules was in the /etc/udev/rules.d/ and that the kinect was appropriately connected to the usb 3.0 port on the pi. 

pi@raspberrypi:~/libfreenect2/build/bin $ lsusb
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 009: ID 045e:097c Microsoft Corp.
Bus 002 Device 010: ID 045e:097d Microsoft Corp.
Bus 002 Device 008: ID 045e:097a Microsoft Corp.
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 012: ID 045e:097e Microsoft Corp.
Bus 001 Device 011: ID 045e:097b Microsoft Corp.
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

When I tried doing an image capture from the kinect using fswebcam, it worked successfully:
pi@raspberrypi:~ $ fswebcam image3.jpg
--- Opening /dev/video0...
Trying source module v4l2...
/dev/video0 opened.
No input was specified, using the first.
Adjusting resolution from 384x288 to 1280x720.
--- Capturing frame...
Captured frame in 0.00 seconds.
--- Processing captured image...
Writing JPEG image to 'image3.jpg'.

I had looked at all the other committed issues and was unable to find a solution among them so far. I do not know what to do next.

floe, 2020-03-31:
Try unloading the kernel module first (`gspca_kinect` IIRC).

yellowj48, 2024-03-02:
I know this was a while ago but did you ever find the solution? I'm running into the same exact issue.