OpenKinect/libfreenect2 #1119: make: *** [all] Error 2

tavishm, 2020-04-16:
Overview Description:

Version, Platform, and Hardware Bug Found:

1. `git log -1 --oneline`
output: `fd64c5d (HEAD -> master, origin/master, origin/HEAD) Update README.md`

2. `uname -a`
output: `fd64c5d (HEAD -> master, origin/master, origin/HEAD) Update README.md`

3. `lsusb -t`
output: 
```
/:  Bus 08.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
/:  Bus 07.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 480M
    |__ Port 3: Dev 2, If 0, Class=Vendor Specific Class, Driver=spca561, 12M
/:  Bus 06.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Vendor Specific Class, Driver=, 5000M
    |__ Port 1: Dev 2, If 1, Class=Vendor Specific Class, Driver=, 5000M
    |__ Port 1: Dev 2, If 2, Class=Audio, Driver=snd-usb-audio, 5000M
    |__ Port 1: Dev 2, If 3, Class=Audio, Driver=snd-usb-audio, 5000M
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 480M
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 10000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/8p, 10000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/14p, 480M
    |__ Port 12: Dev 2, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
    |__ Port 13: Dev 3, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M
    |__ Port 13: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M

```

4. `lspci -nn`
output: 
```
00:00.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Root Complex [1022:1450]
00:01.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) PCIe Dummy Host Bridge [1022:1452]
00:01.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) PCIe GPP Bridge [1022:1453]
00:02.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) PCIe Dummy Host Bridge [1022:1452]
00:03.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) PCIe Dummy Host Bridge [1022:1452]
00:03.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) PCIe GPP Bridge [1022:1453]
00:04.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) PCIe Dummy Host Bridge [1022:1452]
00:07.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) PCIe Dummy Host Bridge [1022:1452]
00:07.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Internal PCIe GPP Bridge 0 to Bus B [1022:1454]
00:08.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) PCIe Dummy Host Bridge [1022:1452]
00:08.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Internal PCIe GPP Bridge 0 to Bus B [1022:1454]
00:14.0 SMBus [0c05]: Advanced Micro Devices, Inc. [AMD] FCH SMBus Controller [1022:790b] (rev 59)
00:14.3 ISA bridge [0601]: Advanced Micro Devices, Inc. [AMD] FCH LPC Bridge [1022:790e] (rev 51)
00:18.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Data Fabric: Device 18h; Function 0 [1022:1460]
00:18.1 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Data Fabric: Device 18h; Function 1 [1022:1461]
00:18.2 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Data Fabric: Device 18h; Function 2 [1022:1462]
00:18.3 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Data Fabric: Device 18h; Function 3 [1022:1463]
00:18.4 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Data Fabric: Device 18h; Function 4 [1022:1464]
00:18.5 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Data Fabric: Device 18h; Function 5 [1022:1465]
00:18.6 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Data Fabric: Device 18h; Function 6 [1022:1466]
00:18.7 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Data Fabric: Device 18h; Function 7 [1022:1467]
00:19.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Data Fabric: Device 18h; Function 0 [1022:1460]
00:19.1 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Data Fabric: Device 18h; Function 1 [1022:1461]
00:19.2 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Data Fabric: Device 18h; Function 2 [1022:1462]
00:19.3 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Data Fabric: Device 18h; Function 3 [1022:1463]
00:19.4 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Data Fabric: Device 18h; Function 4 [1022:1464]
00:19.5 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Data Fabric: Device 18h; Function 5 [1022:1465]
00:19.6 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Data Fabric: Device 18h; Function 6 [1022:1466]
00:19.7 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Data Fabric: Device 18h; Function 7 [1022:1467]
01:00.0 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] X399 Series Chipset USB 3.1 xHCI Controller [1022:43ba] (rev 02)
01:00.1 SATA controller [0106]: Advanced Micro Devices, Inc. [AMD] X399 Series Chipset SATA Controller [1022:43b6] (rev 02)
01:00.2 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] X399 Series Chipset PCIe Bridge [1022:43b1] (rev 02)
02:00.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] 300 Series Chipset PCIe Port [1022:43b4] (rev 02)
02:02.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] 300 Series Chipset PCIe Port [1022:43b4] (rev 02)
02:03.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] 300 Series Chipset PCIe Port [1022:43b4] (rev 02)
[... 45 lines cut ...]
Error caused by step 7:

```
[  4%] Built target generate_resources_tool
[  7%] Building NVCC (Device) object CMakeFiles/cuda_compile_1.dir/src/cuda_compile_1_generated_cuda_kde_depth_packet_processor.cu.o
In file included from /usr/local/cuda/include/host_config.h:50:0,
                 from /usr/local/cuda/include/cuda_runtime.h:78,
                 from <command-line>:0:
/usr/local/cuda/include/crt/host_config.h:119:2: error: #error -- unsupported GNU version! gcc versions later than 6 are not supported!
 #error -- unsupported GNU version! gcc versions later than 6 are not supported!
  ^~~~~
CMake Error at cuda_compile_1_generated_cuda_kde_depth_packet_processor.cu.o.RelWithDebInfo.cmake:219 (message):
  Error generating
  /home/tavishmankash/libfreenect2/build/CMakeFiles/cuda_compile_1.dir/src/./cuda_compile_1_generated_cuda_kde_depth_packet_processor.cu.o

CMakeFiles/freenect2.dir/build.make:82: recipe for target 'CMakeFiles/cuda_compile_1.dir/src/cuda_compile_1_generated_cuda_kde_depth_packet_processor.cu.o' failed
make[2]: *** [CMakeFiles/cuda_compile_1.dir/src/cuda_compile_1_generated_cuda_kde_depth_packet_processor.cu.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/freenect2.dir/all' failed
make[1]: *** [CMakeFiles/freenect2.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

```
Additional Information:
I am using ubuntu 18.04. Please tell me what should I do?

floe, 2020-04-16:
The error message says it all: CUDA only works with gcc 6. So either remove gcc-7 and install gcc-6 instead, or disable CUDA support.

tavishm, 2020-04-17:
Thank you very much for your quick answer. I know this is a stupid question but if I disable CUDA wouldn't it disturb my whole PC and other stuff I've done with CUDA? Or if I remove gcc-7 and install gcc 6  couldn't that damage other dependencies? Is there a way I can disable CUDA only for libfreenect2 stuff? I am not so great at C, so it isn't very easy for me to understand this. Please help me?

floe, 2020-04-17:
Try running cmake with the `-DENABLE_CUDA=OFF` switch. Does that help?

tavishm, 2020-04-17:
I am really thankful to you for this! I had to switch OSs for this. It works now

There is just one problem now, When I run protonect,the images come for some time and then they hang. at the log, I get this error - 
`[Debug] [DepthPacketStreamParser] subpacket too large`

floe, 2020-04-17:
That's a different issue, so let's close this one, OK? Regarding the hangup, this sounds like an USB issue, the only solution would be to switch to a different USB3 controller.

closed 2020-04-17 by floe

floe, 2020-04-17:
P.S. Or maybe tweak some driver/kernel parameters for your USB3 controller, see README.