OpenKinect/libfreenect2 #1121: no device connected!

salvatoreitachi, 2020-05-20:
Hello.I use the Ubuntu16.04 and "clone\cmake\make\make install"already.But when I run the"./bin/Protonect",it didn't work and show"no device connected!"
I did everything I could do but it still didn't work. 
Please help me,thx.

Draal8, 2020-06-09:
The kinect is on a usb 3.0 port on your pc right ? On other usb ports it just won't detect the kinect.

acritarchy, 2020-06-12:
Hi all, I'm getting the same error on 18.04. Working way through troubleshooting guide; can confirm kinect is on USB3.0. 

dmesg after running Protonect returned:

(I bolded the two lines I saw as possible fail conditions)

Any help very appreciated!

```
[ 9977.239476] usb 1-2: USB disconnect, device number 3

[ 9977.239481] usb 1-2.3: USB disconnect, device number 5

[ 9977.991963] usb 2-4: USB disconnect, device number 2

[ 9977.991968] usb 2-4.1: USB disconnect, device number 4

[ 9978.057408] usb 2-4.2: USB disconnect, device number 6

[ 9981.339858] usb 1-2: new high-speed USB device number 11 using xhci_hcd

[ 9981.492295] usb 1-2: New USB device found, idVendor=045e, idProduct=097b, bcdDevice= 1.27

[ 9981.492300] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3

[ 9981.492303] usb 1-2: Product: 4-Port USB 2.0 Hub

[ 9981.492306] usb 1-2: Manufacturer: Generic

[ 9981.492308] usb 1-2: SerialNumber: 000269494612

[ 9981.493617] hub 1-2:1.0: USB hub found

[ 9981.493965] hub 1-2:1.0: 4 ports detected

[ 9981.616147] usb 2-4: new SuperSpeed Gen 1 USB device number 7 using xhci_hcd

[ 9981.639518] usb 2-4: New USB device found, idVendor=045e, idProduct=097a, bcdDevice= 1.27

[ 9981.639523] usb 2-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3

[ 9981.639527] usb 2-4: Product: 4-Port USB 3.0 Hub

[ 9981.639530] usb 2-4: Manufacturer: Generic

[ 9981.639533] usb 2-4: SerialNumber: 000269494612

[ 9981.642285] hub 2-4:1.0: USB hub found

[ 9981.642549] hub 2-4:1.0: 2 ports detected

[ 9981.787857] usb 1-2.3: new high-speed USB device number 12 using xhci_hcd

[ 9981.889044] usb 1-2.3: New USB device found, idVendor=045e, idProduct=097e, bcdDevice= 1.06

[ 9981.889049] usb 1-2.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3

[ 9981.889052] usb 1-2.3: Product: Azure Kinect Microphone Array

[ 9981.889055] usb 1-2.3: Manufacturer: Microsoft Corporation.

[ 9981.889058] usb 1-2.3: SerialNumber: 000269494612

[ 9981.894775] hid-generic 0003:045E:097E.0007: hiddev0,hidraw1: USB HID v1.11 Device 
[Microsoft Corporation. Azure Kinect Microphone Array] on usb-0000:00:14.0-2.3/input2

[ 9981.967907] usb 2-4.1: new SuperSpeed Gen 1 USB device number 8 using xhci_hcd

[ 9981.988670] usb 2-4.1: Isoc endpoint with wBytesPerInterval of 1024 in config 1 interface 1 altsetting 4 ep 129: setting to 944

[ 9981.989020] usb 2-4.1: New USB device found, idVendor=045e, idProduct=097d, bcdDevice= 0.0c

[ 9981.989024] usb 2-4.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3

[ 9981.989027] usb 2-4.1: Product: Azure Kinect 4K Camera

[ 9981.989030] usb 2-4.1: Manufacturer: Microsoft

[ 9981.989032] usb 2-4.1: SerialNumber: 000269494612

[ 9981.992102] uvcvideo: Found UVC 1.00 device Azure Kinect 4K Camera (045e:097d)

**[ 9981.992235] uvcvideo: Failed to query (GET_INFO) UVC control 3 on unit 1: -32 (exp. 1)
.
[ 9981.993813] uvcvideo: Failed to set UVC probe control : -32 (exp. 26).**

[ 9981.997466] uvcvideo 2-4.1:1.0: Entity type for entity Extension 2 was not initialized!

[ 9981.997471] uvcvideo 2-4.1:1.0: Entity type for entity Extension 6 was not initialized!

[ 9981.997474] uvcvideo 2-4.1:1.0: Entity type for entity Extension 5 was not initialized!

[ 9981.997476] uvcvideo 2-4.1:1.0: Entity type for entity Processing 3 was not initialized!

[ 9981.997479] uvcvideo 2-4.1:1.0: Entity type for entity Camera 1 was not initialized!

[ 9981.997765] input: Azure Kinect 4K Camera as /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4.1/2-4.1:1.0/input/input21

[ 9982.076016] usb 2-4.2: new SuperSpeed Gen 1 USB device number 9 using xhci_hcd

[ 9982.096868] usb 2-4.2: New USB device found, idVendor=045e, idProduct=097c, bcdDevice= 1.00

[ 9982.096873] usb 2-4.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3

[ 9982.096876] usb 2-4.2: Product: Azure Kinect Depth Camera

[ 9982.096879] usb 2-4.2: Manufacturer: Microsoft

[ 9982.096882] usb 2-4.2: SerialNumber: 000269494612
```

floe, 2020-06-12:
For Kinect Azure, use https://github.com/microsoft/Azure-Kinect-Sensor-SDK instead of Protonect.

MakerStuff, 2020-12-07:
Hi,
I tried https://github.com/OpenKinect/libfreenect2#linux on Ubuntu 20.04 LTS running the following commands
```
git clone https://github.com/OpenKinect/libfreenect2.git
cd libfreenect2
sudo apt-get install build-essential cmake pkg-config libusb-1.0-0-dev libturbojpeg0-dev libglfw3-dev beignet-dev -y
mkdir build && cd build
cmake .. -DCMAKE_INSTALL_PREFIX=$HOME/freenect2 -Dfreenect2_DIR=$HOME/freenect2/lib/cmake/freenect2
make
make install
sudo cp ../platform/linux/udev/90-kinect2.rules /etc/udev/rules.d/
# replug device
./bin/Protonect
```
`dmesg | grep 045e` returns a lot of "New USB device found", no errors
`[…] usb 2-4.1: New USB device found, idVendor=045e, idProduct=02d9, bcdDevice= 0.73`

`lsusb | grep 045e` returns
```
Bus 002 Device 046: ID 045e:02d9 Microsoft Corp.
Bus 001 Device 085: ID 045e:02d9 Microsoft Corp. 
```

However i can't seem to get the kinect running the way i want it to do.
My full output of `$HOME/libfreenect2/build/bin/Protonect` with and without `sudo` with the kinect attached is
```
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: /home/fabian/libfreenect2/build/bin/Protonect [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 18 usb devices connected
[Info] [Freenect2Impl] found 0 devices
no device connected!
```
The number "18" drops down to 16 when i have the kinect disconnected.

What am i doing wrong here?

thelabcat, 2022-05-09:
I am running various demos, but all have not found my Kinect 2.0. Is definitely plugged into USB 3.0, and the hardware has worked with Windows in the past with the official SDK (now on Linux Mint).