OpenKinect/libfreenect2 #1123: [Error] [VaapiFrame] vaGetImage(display, surface, 0, 0, image.width, image.height, image.image_id): unknown libva error

MengHao666, 2020-06-17:
Overview Description:

when i run **sudo ./bin/Protonect** on ubuntu 20.04  ,I got the error above.

Version, Platform, and Hardware Bug Found:

1. `git log -1 --oneline`
2. `uname -a` 
Linux Lenovo 5.4.0-37-generic #41-Ubuntu SMP Wed Jun 3 18:57:02 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

3. `lsusb -t`
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/8p, 5000M
    |__ Port 2: Dev 13, If 0, Class=Hub, Driver=hub/1p, 5000M
        |__ Port 1: Dev 14, If 0, Class=Vendor Specific Class, Driver=, 5000M
        |__ Port 1: Dev 14, If 1, Class=Vendor Specific Class, Driver=, 5000M
        |__ Port 1: Dev 14, If 2, Class=Audio, Driver=snd-usb-audio, 5000M
        |__ Port 1: Dev 14, If 3, Class=Audio, Driver=snd-usb-audio, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/16p, 480M
    |__ Port 1: Dev 10, If 1, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 1: Dev 10, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 2: Dev 12, If 0, Class=Hub, Driver=hub/1p, 480M
    |__ Port 5: Dev 4, If 1, Class=Video, Driver=uvcvideo, 480M
    |__ Port 5: Dev 4, If 0, Class=Video, Driver=uvcvideo, 480M
    |__ Port 7: Dev 5, If 0, Class=Wireless, Driver=btusb, 12M
    |__ Port 7: Dev 5, If 1, Class=Wireless, Driver=btusb, 12M

4. `lspci -nn`
00:00.0 Host bridge [0600]: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor Host Bridge/DRAM Registers [8086:1910] (rev 07)
00:01.0 PCI bridge [0604]: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor PCIe Controller (x16) [8086:1901] (rev 07)
00:02.0 VGA compatible controller [0300]: Intel Corporation HD Graphics 530 [8086:191b] (rev 06)
00:14.0 USB controller [0c03]: Intel Corporation 100 Series/C230 Series Chipset Family USB 3.0 xHCI Controller [8086:a12f] (rev 31)
00:14.2 Signal processing controller [1180]: Intel Corporation 100 Series/C230 Series Chipset Family Thermal Subsystem [8086:a131] (rev 31)
00:16.0 Communication controller [0780]: Intel Corporation 100 Series/C230 Series Chipset Family MEI Controller #1 [8086:a13a] (rev 31)
00:17.0 SATA controller [0106]: Intel Corporation HM170/QM170 Chipset SATA Controller [AHCI Mode] [8086:a103] (rev 31)
00:1c.0 PCI bridge [0604]: Intel Corporation 100 Series/C230 Series Chipset Family PCI Express Root Port #5 [8086:a114] (rev f1)
00:1c.5 PCI bridge [0604]: Intel Corporation 100 Series/C230 Series Chipset Family PCI Express Root Port #6 [8086:a115] (rev f1)
00:1e.0 Signal processing controller [1180]: Intel Corporation 100 Series/C230 Series Chipset Family Serial IO UART #0 [8086:a127] (rev 31)
00:1f.0 ISA bridge [0601]: Intel Corporation HM170 Chipset LPC/eSPI Controller [8086:a14e] (rev 31)
00:1f.2 Memory controller [0580]: Intel Corporation 100 Series/C230 Series Chipset Family Power Management Controller [8086:a121] (rev 31)
00:1f.3 Audio device [0403]: Intel Corporation 100 Series/C230 Series Chipset Family HD Audio Controller [8086:a170] (rev 31)
00:1f.4 SMBus [0c05]: Intel Corporation 100 Series/C230 Series Chipset Family SMBus [8086:a123] (rev 31)
01:00.0 3D controller [0302]: NVIDIA Corporation GM107M [GeForce GTX 950M] [10de:139a] (rev a2)
02:00.0 Network controller [0280]: Intel Corporation Dual Band Wireless-AC 3165 Plus Bluetooth [8086:3166] (rev 99)
03:00.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller [10ec:8168] (rev

Steps to Reproduce:

1. I instal ubuntu20.04 
2. and I jsut follow the instructions om the site
3. and at last i run   **sudo ./bin/Protonect**

Actual Results:
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 8 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:14 with serial 035566341047
[Info] [Freenect2Impl] found 1 devices
libva info: VA-API version 1.7.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_7
libva info: va_openDriver() returns 0
[Info] [VaapiRgbPacketProcessorImpl] driver: Intel iHD driver for Intel(R) Gen Graphics - 20.1.1 ()
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9855
[Debug] [Freenect2DeviceImpl] status 0x090000: 9855
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 035566341047
device firmware: 4.0.3917.0
[Debug] [DepthPacketStreamParser] not all subsequences received 0
[Error] [VaapiFrame] vaGetImage(display, surface, 0, 0, image.width, image.height, image.image_id): unknown libva error
[Debug] [DepthPacketStreamParser] not all subsequences received 512
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
Segmentation fault

Expected Results:
I want to get the RGB Depth data
Reproducibility:

Additional Information:
Could u make a new branch for newer system

SiddhantNadkarni, 2020-07-22:
Same error. Any updates on this?

PatricianEagle, 2020-12-05:
Try these commands.

gedit ~/.bash_profile
export LIBVA_DRIVER_NAME=i965
source ~/.bash_profile

haider012829, 2021-10-13:
> Same error. Any updates on this?

Have you solved your problem?

yuukireina05, 2021-11-03:
For me I uninstall all packages with name 'libva' except for libva-drm2 it worked. Seems that I installed Wine for my Ubuntu and it brought about some unexpected packages that causing this crash.

remmel, 2022-02-16:
Uninstalling libva-dev like @Minionflo say. Rerun cmake (it will disable VAAPI) and make

fahnub, 2024-12-03:
thanks @remmel @Minionflo 
uninstalling libva-dev and rerunning cmake and make worked for me