OpenKinect/libfreenect2 #1131: RGB image squished to 1/3 size

rcslaney, 2020-10-29:
Hi there,

I have a problem with the RGB data coming from the Kinect. It appears that the RGB data is squished to 1/3 of the image frame both in Protonect (as shown in the screenshot below) and when trying to use other example programs.

How can I fix this? I know I could crop and scale it but then I'm getting 1/3 of the intended resolution horizontally.

Many thanks

[image]

marekjg, 2022-11-14:
Hi, I've finally fixed the problem. Run `Protonect` and search for line with `VaapiRgbPacketProcessorImpl`, e.x. `[Info] [VaapiRgbPacketProcessorImpl] driver: Intel iHD driver for Intel(R) Gen Graphics - 22.3.1`. For me this driver was producing squished image.
To change the driver, check [VA-API docs](https://wiki.archlinux.org/title/Hardware_video_acceleration#Configuring_VA-API) and `ls /usr/lib/x86_64-linux-gnu/dri/`.

leocornelius, 2023-03-10:
Hi there, I’m having the same issue. I’m using my laptop as my pc’s usb 3 ports are unsupported. It has an intel I3 and is running Ubuntu 22.04. It defaults to using iHD and it initialises that fine (ie it says initialising returns 0/ok) but results in the same issue you described. If I change the driver used to I965 the issue goes away and the output is as expected and without any performance issues BUT I then looked at the console and realised that it was only working because vappi wouldn’t initialise (I can’t remember the exact error off the top of my head and not at home right now) and so protonect would just not use it. 

If it runs just fine without vaapi then can I disable vaapi use through code? Doing so by just forcing it to crash seems a little hacky.

but I’m unsure if this is really a solution, and would rather try and fix the issue with the iHD driver. Perhaps someone with more understanding on the workings of the libary and what it uses vaapi for can shed some light as to why iHD causes such weird problems. It might be the case that just not using vaapi is the easiest solution.