OpenKinect/libfreenect2 #1140: unknown libva error

Mrsirovo, 2020-12-13:
Overview Description:
When I try to run `./bin/Protonect gl`
I got 
`Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
libva info: VA-API version 1.7.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_7
libva info: va_openDriver() returns 0
[Info] [VaapiRgbPacketProcessorImpl] driver: Intel iHD driver for Intel(R) Gen Graphics - 20.1.1 ()
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 7 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:4 with serial 007198363647
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9855
[Debug] [Freenect2DeviceImpl] status 0x090000: 9855
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 007198363647
device firmware: 4.0.3916.0
[Debug] [DepthPacketStreamParser] not all subsequences received 0
[Error] [VaapiFrame] vaGetImage(display, surface, 0, 0, image.width, image.height, image.image_id): unknown libva error
[Debug] [DepthPacketStreamParser] not all subsequences received 1008
段错误 (核心已转储)`

Version, Platform, and Hardware Bug Found:

1. `git log -1 --oneline`
2. `uname -a`
Linux mrsir-Lenovo 5.4.0-56-generic #62-Ubuntu SMP Mon Nov 23 19:20:19 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

3. `lsusb -t`
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/8p, 10000M
    |__ Port 2: Dev 4, If 2, Class=Audio, Driver=snd-usb-audio, 5000M
    |__ Port 2: Dev 4, If 0, Class=Vendor Specific Class, Driver=, 5000M
    |__ Port 2: Dev 4, If 3, Class=Audio, Driver=snd-usb-audio, 5000M
    |__ Port 2: Dev 4, If 1, Class=Vendor Specific Class, Driver=, 5000M
    |__ Port 3: Dev 3, If 0, Class=Mass Storage, Driver=uas, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/16p, 480M
    |__ Port 1: Dev 2, If 1, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 1: Dev 2, If 2, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 1: Dev 2, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 9: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 14: Dev 4, If 0, Class=Wireless, Driver=btusb, 12M
    |__ Port 14: Dev 4, If 1, Class=Wireless, Driver=btusb, 12M

4. `lspci -nn`
00:00.0 Host bridge [0600]: Intel Corporation Device [8086:9b44] (rev 02)
00:01.0 PCI bridge [0604]: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor PCIe Controller (x16) [8086:1901] (rev 02)
00:02.0 VGA compatible controller [0300]: Intel Corporation UHD Graphics [8086:9bc4] (rev 05)
00:04.0 Signal processing controller [1180]: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor Thermal Subsystem [8086:1903] (rev 02)
00:08.0 System peripheral [0880]: Intel Corporation Xeon E3-1200 v5/v6 / E3-1500 v5 / 6th/7th/8th Gen Core Processor Gaussian Mixture Model [8086:1911]
00:12.0 Signal processing controller [1180]: Intel Corporation Comet Lake PCH Thermal Controller [8086:06f9]
00:14.0 USB controller [0c03]: Intel Corporation Comet Lake USB 3.1 xHCI Host Controller [8086:06ed]
00:14.2 RAM memory [0500]: Intel Corporation Comet Lake PCH Shared SRAM [8086:06ef]
00:14.3 Network controller [0280]: Intel Corporation Wi-Fi 6 AX201 [8086:06f0]
00:15.0 Serial bus controller [0c80]: Intel Corporation Comet Lake PCH Serial IO I2C Controller #0 [8086:06e8]
00:15.1 Serial bus controller [0c80]: Intel Corporation Comet Lake PCH Serial IO I2C Controller #1 [8086:06e9]
00:16.0 Communication controller [0780]: Intel Corporation Comet Lake HECI Controller [8086:06e0]
00:17.0 SATA controller [0106]: Intel Corporation Device [8086:06d3]
00:1b.0 PCI bridge [0604]: Intel Corporation Comet Lake PCI Express Root Port #17 [8086:06c0] (rev f0)
00:1d.0 PCI bridge [0604]: Intel Corporation Comet Lake PCI Express Root Port #9 [8086:06b0] (rev f0)
00:1d.6 PCI bridge [0604]: Intel Corporation Device [8086:06b6] (rev f0)
00:1f.0 ISA bridge [0601]: Intel Corporation Comet Lake LPC Controller [8086:068d]
00:1f.3 Audio device [0403]: Intel Corporation Comet Lake PCH cAVS [8086:06c8]
00:1f.4 SMBus [0c05]: Intel Corporation Comet Lake PCH SMBus Controller [8086:06a3]
00:1f.5 Serial bus controller [0c80]: Intel Corporation Comet Lake PCH SPI Controller [8086:06a4]
01:00.0 VGA compatible controller [0300]: NVIDIA Corporation Device [10de:1f99] (rev a1)
01:00.1 Audio device [0403]: NVIDIA Corporation Device [10de:10fa] (rev a1)
06:00.0 Non-Volatile memory controller [0108]: Phison Electronics Corporation Device [1987:5013] (rev 01)
07:00.0 Non-Volatile memory controller [0108]: Samsung Electronics Co Ltd NVMe SSD Controller SM981/PM981/PM983 [144d:a808]
08:00.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller [10ec:8168] (rev 15)

I have tried the following:
gedit ~/.bash_profile
export LIBVA_DRIVER_NAME=i965
source ~/.bash_profile

But it still doesn't work.

parkerlreed, 2021-01-11:
Same here

Latest git with 

```
libva 2.10.0-1
libva-mesa-driver 20.3.2-2
```

```
[parker@wolfcola bin]$ Protonect
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: Protonect [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 8 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:4 with serial 067713440247
[Info] [Freenect2Impl] found 1 devices
[Info] [VaapiRgbPacketProcessorImpl] driver: Mesa Gallium driver 20.3.2 for AMD Radeon (TM) RX 480 Graphics (POLARIS10, DRM 3.40.0, 5.10.5-zen1-1-zen, LLVM 11.0.0)
[Warning] [VaapiRgbPacketProcessorImpl] YUV444 not supported by libva, chroma will be halved
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9763
[Debug] [Freenect2DeviceImpl] status 0x090000: 9763
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 067713440247
device firmware: 4.0.3916.0
[Debug] [DepthPacketStreamParser] not all subsequences received 0
[Error] [VaapiFrame] vaGetImage(display, surface, 0, 0, image.width, image.height, image.image_id): operation failed
Segmentation fault (core dumped)
```

parkerlreed, 2021-01-11:
Oh good what's old is new again. I've posted the same thing before

https://github.com/OpenKinect/libfreenect2/issues/1026#issuecomment-578161074

parkerlreed, 2021-06-05:
Any updates here? Would love to be able to use VAAPI drivers outside of Intel.

parkerlreed, 2023-04-20:
Any chance of getting libva support for AMD? It has been well supported for years now.