OpenKinect/libfreenect2 #1144: Can I use the Kinect as a camera device?

vncscoelho, 2021-01-27:
I'm trying to make my Macbook with Big Sur to recognize the Kinect as a camera device, so I can use it as a webcam. Installation went well and Protonect runs without errors.

Thanks

Beyley, 2021-05-01:
Im currently trying to do this on Pop os 20.10, Protonect runs fine but im unable to get it working as a camera device

Fusseldieb, 2021-08-18:
I don't really know if beaming IR on your face all day is doing any good. I doubt so tbh...

Beyley, 2021-08-18:
> Unrelated, but...
> 
> I don't really know if beaming IR onto your eyes all day is doing any good. I doubt so tbh...
> 
> If there's a way to disable the IR array, it would be better.

doesn't the sun emit IR? also dont a lot of old lights do? im not sure a kinect is gonna do much, and I've been using it as a webcam for 2 months now with no issues

Fusseldieb, 2021-08-19:
> doesn't the sun emit IR? 

That's why you shouldn't look at it (also because UV)

> im not sure a kinect is gonna do much, and I've been using it as a webcam for 2 months now with no issues

Well, I'm no specialist in this area, so idk, just chiming in ahahaha

Beyley, 2021-08-23:
> > doesn't the sun emit IR?
> 
> That's why you shouldn't look at it (also because UV)
> 
> > im not sure a kinect is gonna do much, and I've been using it as a webcam for 2 months now with no issues
> 
> Well, I'm no specialist in this area, so idk, just chiming in ahahaha

Just looked it up and IR doesnt really effect the human eye, its the UV in the sun that causes the issue (and just how bright it is in the visible spectrum)