OpenKinect/libfreenect2 #1149: run error

xiaoyuamw, 2021-04-21:
Hi,
When i run the command line "sudo ./bin/Protonect", i have met such errors as bellows:
..........................................................................................................................................
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 13 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @6:13 with serial 014215164647
[Info] [Freenect2Impl] found 1 devices
[Error] [OpenGLDepthPacketProcessorImpl] GLFW error 65544 X11: RandR gamma ramp support seems broken
[Error] [VaapiRgbPacketProcessorImpl] vaDisplayIsValid(display) failed
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9729
[Debug] [Freenect2DeviceImpl] status 0x090000: 9731
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Debug] [DepthPacketStreamParser] not all subsequences received 0
[Info] [Freenect2DeviceImpl] started
device serial: 014215164647
device firmware: 4.0.3916.0
[Debug] [DepthPacketStreamParser] skipping depth packet
failed to compile vertex shader!
0:1(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.40, 1.00 ES, and 3.00 ES

failed to compile fragment shader!
0:1(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.40, 1.00 ES, and 3.00 ES

failed to link shader program!
error: linking with uncompiled/unspecialized shadererror: linking with uncompiled/unspecialized shader
failed to compile vertex shader!
0:1(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.40, 1.00 ES, and 3.00 ES

failed to compile fragment shader!
0:1(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.40, 1.00 ES, and 3.00 ES

failed to link shader program!
error: linking with uncompiled/unspecialized shadererror: linking with uncompiled/unspecialized shader
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
'''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''''
How can i solved it?
Thank you!

Best Regards

SirDifferential, 2021-04-23:
> failed to compile vertex shader!
>0:1(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.40, 1.00 ES, and 3.00 ES

That means your GPU driver or GPU hardware does not support GLSL 3.30. What hardware are you using?

Also:

>sudo ./bin/Protonect

You shouldn't be running protonect as root. Set the [udev rules](https://github.com/OpenKinect/libfreenect2/blob/master/README.md#linux) as specified by the README.

xiaoyuamw, 2021-04-23:
HI, @SirDifferential
Thank you for you reply very much!
My GPU is Nvidia 2060, but i hadn't installed the GPU driver before, because i don't want to use it. Meanwhile  when i run the command line in another computer, which can work successfully and it have no GPU!
What should i do next?

Best Regards.

SirDifferential, 2021-04-23:
Which OS and CPU do you have?

Maybe you also have an integrated GPU and Protonect is launching using it. On Windows you can configure program launch settings in the NVidia control panel and specify protonect to be started using the Nvidia GPU. GLSL 3.30 requires at least OpenGL 3.3 support in the driver.

xiaoyuamw, 2021-04-23:
> Which OS and CPU do you have?
>
> Maybe you also have an integrated GPU and Protonect is launching using it. On Windows you can configure program launch settings in the NVidia control panel and specify [...]

My OS is ubuntu18.04 and cpu is AMD 4800H, then how to deal with it?

SirDifferential, 2021-04-23:
4800H includes a Vega series GPU in it. You can use `glxinfo  |grep "version"` to get a list of versions supported by the driver, and `glxinfo |grep "OpenGL vendor"` to see which GPU it's listing the data for.

Other than that, you'll have to find out a way to either:

a) get a driver that supports OpenGL 3.3. I'm not sure how much OpenGL is implemented for Vega GPUs on Linux.
b) launch protonect, somehow, using the Nvidia GPU