OpenKinect/libfreenect2 #1151: How do I use Kinect v2 as webcam after installing LibFreeNect2?

JadnoABS, 2021-06-14:
I'm using Manjaro Linux and I want to use my XBOne Kinect as a webcam device.
I have installed Libfreenect2 and the Kinect works when I run Protonect.

doctorstoo, 2021-06-27:
Please install OpenCV and try streamer_recorder function following this:

### Linux

* Install build tools
```
sudo apt-get install opencv3
```

* Build (start from the libfreenect2 root directory)
```
mkdir build && cd build
cmake .. -DBUILD_STREAMER_RECORDER=ON
make
make install
```

* Run the test program (accepts all the same options as Protonect with 3 extra):
    - `./bin/ProtonectSR -record` -- to start recording enable frames (`freenect2-record` presupposes this option)
    - `./bin/ProtonectSR -stream` -- to start streaming frames to a receiver application (`freenect2-stream` presupposes this option)
    - `./bin/ProtonectSR -replay` -- to start replaying recorded frames (`freenect2-replay` presupposes this option)
    - `./bin/ProtonectSR -replay -stream` -- to relay and stream recorded frames
    - `./bin/ProtonectSR -record -stream` -- to record and stream frames

JadnoABS, 2021-07-02:
Thanks, I installed opencv3 and ran `cmake .. -DBUILD_STREAMER_RECORDER=ON` but when I ran `make` I received several errors like this one:

`/home/user/libfreenect2/tools/streamer_recorder/include/PracticalSocket.h:336:49: error: ISO C++17 does not allow dynamic exception specifications
  336 |   void leaveGroup(const string &multicastGroup) throw(SocketException);`

doctorstoo, 2021-07-06:
Hope this will help you.
https://github.com/r9y9/pylibfreenect2/blob/master/examples/multiframe_listener.py

JadnoABS, 2021-07-06:
I ran that script and it opened a window which showed the correct kinect capture, but still there is no option for kinect on software that uses webcam.

obiwan1337, 2021-12-08:
i came as far as this goes. With 4 individual windows for ir, color, depth and combined input now on my screen I have no idea what do do or how to access those streams. 
If you think about using the virtual camera option of OBS for example you can emulate the webcam and use a recording as input to show.  You'll need the streamwindow to be open all the time with this construction...

> I ran that script and it opened a window which showed the correct kinect capture, but still there is no option for kinect on software that uses webcam.

thelabcat, 2022-05-09:
Another option is to screen capture the Protonect window, and crop it. Kind of clumsy, but works.