OpenKinect/libfreenect2 #1153: Release new version

sh0, 2021-08-05:
Would it be possible to do a new version release? The 0.2.0 does not build with recent GCC compilers, but master branch does. Last release was in 2016 and while there has been little activity in past year due to demise of Kinect 2, there are still some legacy projects using it.

@xlz do you have time to do it?

xlz, 2021-08-05:
A release implies quite a bit of testing and packaging on several platforms. Do you want a source only "release" that basically says "please use master"?

sh0, 2021-08-06:
Yea, just a new git tag 0.2.1 or similar and a tarball of source would suffice.

I want to add this library to a package manager, but git clone is something that should be avoided there. Also patching all changes on top of 0.2.0 is something that I would prefer to avoid.

xlz, 2021-08-06:
https://github.com/OpenKinect/libfreenect2/releases/tag/v0.2.1

closed 2021-08-06 by xlz