OpenKinect/libfreenect2 #1159: Segmentation fault (core dumped)

CivilizedGuide, 2021-10-05:
Hi, I have a problem trying to use libfreenect, especially when I try to use `./Protonect`, it shows me the following error

```
[Error] [VaapiFrame] vaGetImage(display, surface, 0, 0, image.width, image.height, image.image_id): unknown libva error
Segmentation fault (core dumped)
```
my operating system is ubuntu 20.04 and i have an intel hd graphics 520

I do not have much knowledge in programming so I have not solved this problem

Sorry if you don't make me understand, I'm using a translator

GiovanniBalestrieri, 2022-04-02:
I have the same issue. In my case, this is the output of `./bin/Protonect`:

```
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 6 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:2 with serial 040772134647
[Info] [Freenect2Impl] found 1 devices
libva info: VA-API version 1.7.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_7
libva info: va_openDriver() returns 0
[Info] [VaapiRgbPacketProcessorImpl] driver: Intel iHD driver for Intel(R) Gen Graphics - 20.1.1 ()
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9763
[Debug] [Freenect2DeviceImpl] status 0x090000: 9763
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
...
device firmware: 4.0.3912.0
[Debug] [DepthPacketStreamParser] not all subsequences received 0
[Error] [VaapiFrame] vaGetImage(display, surface, 0, 0, image.width, image.height, image.image_id): unknown libva error
Segmentation fault (core dumped)
```
Same output using cpu or gpu options. The only one that doesn't not show the error immediately is 
`./bin/Protonect cl`
And shows
`[Error] [DepthPacketStreamParser] Packet buffer is NULL`

I ll test the sensor on another pc. Just to make sure it is not the root of the problem.

GiovanniBalestrieri, 2022-04-05:
Tried with another USB cable. Same error. I checked my graphic card info using:
```
userk@histamine:~/development/catkin_ws_ros_one$ lspci | grep VGA
00:02.0 VGA compatible controller: Intel Corporation Skylake GT2 [HD Graphics 520] (rev 07)
```
Then `export LIBVA_DRIVER_NAME=i965` to match the Intel graphics driver as suggested by @xlz in [this](https://github.com/OpenKinect/libfreenect2/issues/1026) issue