OpenKinect/libfreenect2 #1162: How do I use the Kinect as a webcam MacOs?

cleggorz, 2021-12-27:
I have successfully installed the driver. Running the ./bin/Protonect shows 4 outputs but the camera is not shown in MacOs as a webcam device. The camera is not detected in MacOs native apps or software like OBS studio. The Xbox Nui sensor is shown under USB 3.0 devices in system information. Am I missing something??

cleggorz, 2021-12-28:
So the only way I have been able to get this to work is a complete mess and probably is not worth it. This project does seem pretty dead though so I guess this is the only way. 

I have used Obs studio to copy the output from Protonect by capturing the viewer. Then in Obs I have cropped the viewer to size and stretched it to the correct size then turned on Obs virtual camera. [image]

smokhov, 2022-03-07:
The other way you'd have to use Syphon on MacOS to share the frames from Protonect to OBS Studio. The latter has Syphon as one of the input sources.

arekhn, 2022-11-04:
> The other way you'd have to use Syphon on MacOS to share the frames from Protonect to OBS Studio. The latter has Syphon as one of the input sources.

Do you think you could make a small guide to achieve this? I have installed libfreenect2 and it works, but I dont understand how to integrate this with Syphon, do I have to install Openni2 too?