OpenKinect/libfreenect2 #1164: Connecting kinect v2 to rpi 4b

emirhannbozkurt, 2022-01-02:
Hello, I am new to Rpi Model 4B and kinect sensor.

For the Kinect v2 I use, I saw that I needed libfreenect2 in projects via python. However, I couldn't find enough resources to connect the sensor to rpi 4b, and I'm a novice. Is there anyone who can help me?

eclabs007, 2022-04-07:
https://rjw57.github.io/freenect2-python/

emirhannbozkurt, 2022-04-07:
> https://rjw57.github.io/freenect2-python/

Hello @eclabs007. This source you shared looks like libfreenect2 installation for windows operating system. Did I misunderstand?

Sandesh4242, 2022-04-14:
> > https://rjw57.github.io/freenect2-python/
> 
> Hello @eclabs007. This source you shared looks like libfreenect2 installation for windows operating system. Did I misunderstand?

No this works in python in Raspberry pi i just tested it